grbl/grbl #863: COREXY X Problem

tiago1080550, 2015-12-02:
Greetings,

I'm having an interesting problem with the movement along the X axis in a COREXY structure (no z-axis).
The homing instruction executes correctly, but on the wrong direction, so i invert it by using $3=3 (the two limit switches are placed in the down left corner).
If after to the homing, I send an movement instruction for only the Y axis (G90G1Y20F500), it's executed without a problem. The error occurs when the same instruction is sent for the X axis, this time the axis does not stop and goes until it slams against the structure and i reset the arduino.
This happens even on a fresh install of GRBL 0.9j, and the interesting thing is I stopped this project for a month and everything was working fine and never had this problem. Am I missing something?

Thanks in advance

chamnit, 2015-12-02:
@tiago1080550 : There is an issue with homing and CoreXY that I've been meaning to fix for a long time. I keep getting pulled away with other tasks and forget. Tehre is another thread that has a solution. I recall it's just a few lines of code to alter to fix it.

tiago1080550, 2015-12-02:
Thanks for your reply, you mean the fix found here -> #818?

Yes, I had it applied. 
The homing runs fine, the problem is when I give an order for the X axis to move and I have the homing dir mask configured to invert the X axis ($23), in this case it won't stop at the specified position, it will run until it slams the structure.

chamnit, 2015-12-02:
@tiago1080550 : In that case, you can enable limit switch state reporting by setting $10=255. This will enable all real-time reports including limit state. Connect to Grbl via a terminal and send '?' character to directly view the reports. Check if your switches are being read by Grbl correctly. 0=not triggered. 1=triggered.

tiago1080550, 2015-12-02:
I think I'm not being specific in the description of my problem, and I am sorry about that.
The limit switches are read correctly, and the homing cicles are completed:

#define HOMING_CYCLE_0 (1<<Y_AXIS)
#define HOMING_CYCLE_1 (1<<X_AXIS)

The problem is after that, in the normal operation of my project. Any movement only in the Y axis is correctly executed but not on the X axis.

chamnit, 2015-12-02:
@tiago1080550 : Ok. It's likely that the origin isn't set correctly are homing, or you're assuming that homing will set the origin wherever it homes to, which Grbl doesn't. Check your machine position after homing to see if it looks correct. And try issuing a G91 incremental motion to see if the x-axis will move short incremental distances.

tiago1080550, 2015-12-02:
Did just now a test, and I you are absolutly correct, using G91 worked as expected. And yes I was assuming homing sets the origin. A G10 solved the issue.

Thank you very much for your help

closed 2015-12-02 by chamnit