grbl/grbl #864: Docs on loading grbl as library are wrong

mabushey, 2015-12-03:
1. Download the Grbl source code (v0.9i). 
```
╭─michael@corwin ~/Arduino/grbl  ‹master› 
╰─➤  git branch -v
* master d226555 Minor bug fixes.
```

2. Make sure you are using the most recent version of the Arduino IDE
```
╭─michael@corwin ~/Arduino/grbl  ‹master› 
╰─➤  pacman -Q arduino
arduino 1:1.6.5.r5-2
```

3. Load Grbl into the Arduino IDE as a Library.
  Sketch -> Include Library -> Add .ZIP library...
  Select /home/michael/Arduino/grbl/grbl (This is the folder that contains the .c and .h files) -> OK

```
Arduino: 1.6.5 (Linux), Board: "Arduino/Genuino Uno"

java.io.FileNotFoundException: /home/michael/Arduino/grbl/grbl/grbl (No such file or directory)
	at java.util.zip.ZipFile.open(Native Method)
	at java.util.zip.ZipFile.<init>(ZipFile.java:220)
	at java.util.zip.ZipFile.<init>(ZipFile.java:150)
	at java.util.zip.ZipFile.<init>(ZipFile.java:164)
	at processing.app.tools.ZipDeflater.<init>(ZipDeflater.java:26)
	at processing.app.Base.handleAddLibrary(Base.java:2478)
	at processing.app.Base$11.actionPerformed(Base.java:1179)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
/home/michael/Arduino/grbl/grbl/grbl (No such file or directory)
```
It is correct, there is no directory `/home/michael/Arduino/grbl/grbl/grbl`. I specified `/home/michael/Arduino/grbl/grbl`. As it likes to add an extra `grbl` I tried using just `/home/michael/Arduino/grbl` but that reports `A subfolder of your sketchbook is not a valid library`

mabushey, 2015-12-03:
Trying to write non Arduino firmware using the Arduino IDE seems like a really dumb idea anyway...

Using Arch, to run `make grbl.hex` you need to install `avr-libc` as well as `avr-gcc`: `pacman -S avr-gcc avr-libc`.

mabushey, 2015-12-03:
Also telling avrdude to use a conf file that no one has and that is not needed because the options are already provided on the CLI is rediculous. Most likely there were lots of changes and the docs were never updated. The avrdude command that worked for me is:
`avrdude -pm328p -carduino -P/dev/ttyACM0 -D -Uflash:w:grbl.hex`

chamnit, 2015-12-03:
@mabushey : I'm sorry you're having problems, and you don't have to be rude. But I think you're missing the point entirely. Using the Arduino IDE to install Grbl is simply a method to make it accessible, as most people know how to use it, and requires no command line. For more advanced users, you can still compile and flash via the outline methods in the Wiki by using the compiler tools in the Arduino IDE itself. Or, compile by installing the tools yourself, as you have probably have done.

The Arduino team has been recently changing how the IDE does things in the backend, and the Grbl installation hack may have been broken. The last I checked everything was still working fine, but if you are indeed having problems, we'll need to update the installing process.

chamnit, 2015-12-03:
Just for closure before heading to bed. I downloaded the current stable build of the Arduino IDE (v1.6.6). Added Grbl source directory with the "Add .ZIP library" menu item, and brought up the GrblUpload example. Grbl successfully compiled and installed, as it should.

I'm not sure what is going wrong, but the `A subfolder of your sketchbook is not a valid library` error is probably a clue. My guess is that the grbl source folder is in some kind of Arduino IDE file structure that is expecting something else. I would move it elsewhere and try again.

109JB, 2015-12-03:
Under Windows 7 and Arduino IDE 1.6.6, the directions on the WIKI page work perfectly. Perhaps it is something weird about the Linux version of the IDE.  Also, at least in the windows version of the IDE you can simply place the grbl folder containing the .c and .h files into the arduino library folder (documents\Arduino\libraries) and then restart the IDE

As for:
"Trying to write non Arduino firmware using the Arduino IDE seems like a really dumb idea anyway..."

This is a pretty harsh statement considering that you are trying to load a free open source firmware that is worked on by people who donate their time to the project. The reason for the WIKI instructions to use the IDE to upload is so that non-programmers can have an easier time of installing the firmware. Incidentally, it is a method that many before you have had no problem with. If you do find a problem, by all means report it but there is no need to call what is being done "dumb" or "ridiculous"

cri-s, 2015-12-03:
I can give it a tray today on linux sind fix it if there are issues.

mabushey, 2015-12-03:
@chamnit - I apologize my frustrations came across as rudeness. That was not the intent. The fact that I feel that using the Arduino IDE to flash GRBL as "dumb" when we have avrdude should not affect you. 

I think you're right that I missed the point. My point is the Arduino IDE gave a crap pile of errors and didn't work. The reason I had problems is the docs are so bad that I found them to be more of a hindrance than a help. I posted what worked for me on a recent Linux system so hopefully the docs could be cleared up. I think all references to avrdude.conf should be removed. Why create and specify a config file when the config options are already specified/overridden on the command line?

I appreciate all the responses. GRBL changes things. It's amazing. Let's make it easier for people to flash it.

chamnit, 2015-12-03:
@mabushey : The "avrdude.conf" is required (at least on my Mac) when compiling Grbl using the Arduino IDE's avrgcc and other tools. I would imagine you wouldn't need it, if you have avr tools already install separately on your machine. Again, using the Arduino IDE or the IDE's backend tools directly via command line provides a consistent basis between OS-types and versions. It makes it much easier to support.

Grbl's docs are Wiki-based, so if you have suggestions on how to fix them, please feel free to make edits or additions so that others don't run into problems as you did.

mabushey, 2015-12-04:
@chamnit  I have Arch, Debian, Ubuntu, and FreeBSD machines. I'll do some testing this weekend and see if I can clean up the docs. M

mabushey, 2015-12-08:
I was able to flash grbl on a Ubuntu machine in less than 5 minutes start to finish using avrdude. I'm thinking about creating a docker image that will automatically flash GRBL on an Uno.

kyberkiller, 2016-01-07:
Is there any solution to the problem itself? The problem persists in Ubuntu 14.04 LTS at least.