grbl/grbl #865: Grbl 0.9 + 5 relay outputs

mojalovaa, 2015-12-18:
Hi
I use GRBL + MEGA2560  for my diy cnc machine and need add M commands and options for ON/OFF relay with some command , can  be M or some other .
Can some one explain me where all need make changes  and for sample  write me  how need look that code for one  relay?
I m newer  before  work development  software .
On this moment I use spindle and coolant options for relay  , but need min. one more relay outputs and on spindle CW and CCW  have invert situations or something like that .
My plane is make something like this :
Relay 1 (for sample ) M300 (On ) M305  (Off )
Relay 2 ( for sample ) M301 (On) M306  (Off )
Relay 3 (for sample ) M302 (On ) M307  (Off )
Relay 4 ( for sample ) M303 (On) M308  (Off )
Relay 5 (for sample ) M304 (On ) M309  (Off )

Soy for my disaster english .

chamnit, 2016-01-11:
@mojalovaa : "gcode.c" is where the M-codes are parsed and executed. I would mimic the spindle_control.c and coolant_control.c files for the execution of your relays. It's all very well commented.

closed 2016-01-11 by chamnit

gecrby, 2016-03-06:
У меня получилось для четырех реле,но пока "M0" не срабатывает, не знаю почему

http://www.cnc-club.ru/forum/viewtopic.php?f=151&t=7459&p=260998#p260998
 grbl_PU20160223.zip

gecrby, 2016-03-06:
Мой старший сын наблюдая за моим творчеством сделал мне подарок, 4 реле. Мне удалось убедить его опубликовать эти дополнения.

https://github.com/pusso/grbl

pusso, 2016-03-06:
Hello, my commit requared 2560 processor in config.h
// Default cpu mappings. Grbl officially supports the Arduino Uno only. Other processor types
// may exist from user-supplied templates or directly user-defined in cpu_map.h
//#define CPU_MAP_ATMEGA328P // Arduino Uno CPU
#define CPU_MAP_ATMEGA2560 // Arduino MEGA 2560 CPU

Thanx for all.

luisluna2, 2018-08-28:
hello somebody know how add this 5 relay outputs in GRBL 1.1f?