grbl/grbl #867: Z Axis not homing - XY home fine

smartroad, 2015-12-20:
Having an issue with homing the Z axis. Using 0.9j. When I hit home the Z axis (which is off of the limit switch) moves up a small amount - then moves down - up slowly then down a small amount then stops. Exactly the behaviour I would expect it to do if it hit the limit switch but it is no where near it.

The X and Y axis home exactly as it should, move over, click the limit switch, slow move and then pull off 1mm. The Z axis just doesn't move up towards the limit switch. If I hold the limit switch (pretending the Z axis is already home) it throws an error after moving down a little: "ALARM: Homing fail".

I am guessing I have a setting (software or physical) wrong somewhere but no idea where to look as I am confused due to the XY homing fine.

chamnit, 2015-12-20:
@smartroad : It sounds like you have everything wired and setup correctly. The problem you are seeing is most likely an issue with electric noise in your z-limit wire. It's falsely triggering the z-limit pin during the z-homing cycle. Try disconnecting the z-limit switch at the Arduino and jump it manually to rule the wire out. If it is that, wire in a low-pass filter with a small capacitor.

smartroad, 2015-12-20:
Your sir are amazing! 100nF between pin and GND and Z is homing nicely! Many thanks for your time :)

closed 2015-12-20 by smartroad

davemaster, 2015-12-21:
Try this:

1- Check the z-stop sensor
2- Chenk in firmware if the z-stop is inverse/no-inverse mode according your printer model

nkavinash, 2020-02-19:
i have my homebrewd cnc mill, intially tested for the functions like homming cycle and was behaving as it should, like mooving up Z first, click n away from limit swithc, then XY mooves click n moove away, mean while my pppppc had prob n needed the harddisk swap, now with only  Z homes up click n pull away,the XY, does not moove, kindly help , i am lay man n just an enthusiast,

Mamu05, 2021-09-02:
> Your sir are amazing! 100nF between pin and GND and Z is homing nicely! Many thanks for your time :)

I am having an identical issue. Do you think you could please detail your resolve so I can imitate what you did to fix the problem?

rafahikenfly, 2021-09-29:
> > Your sir are amazing! 100nF between pin and GND and Z is homing nicely! Many thanks for your time :)
> 
> I am having an identical issue. Do you think you could please detail your resolve so I can imitate what you did to fix the problem?

@Mamu05, I just faced the same problem. My CNC machine should home Z axis but it moved to feed some millimeters, then it seek some more millimeters as if it was triggering the end stop switch instead. I fixed the issue as follows:
a. Made sure my switch was working properly, by check-in continuity in the hardware.
b. Made sure my wire was working properly, by manually triggering it and checking Grbl status (send ? and look for an end-stop indicator in the status message)
c. Made sure my Z motor was working properly, by moving it up and down both long and short movements, different feed rates.

Since everything was working fine, I got a 104 (100nF) capacitor and placed it between the z end-stop pins. I put it directly in the cable`s dupont terminal on the Arduino side (I'm using a v.3 CNC Shield), no soldering was needed yet. It solved the electric noise issue (my z end-stop cable is about 5m long).

Hope this help you!