grbl/grbl #869: GRBL + X, Y motors + Servo ( on Arduino Uno )

DBSDBS, 2015-12-22:
Hi,

Is it possible to do control X, Y motors + Servo ( for a shutter for the projector)  on Arduino Uno?

M03 Sxx / M04 Sxx do some moves, but is it possible to use them correctly? 

Thanks in advance.

mat60, 2015-12-22:
Hi, 

You can try with grbl called grbl-servo : https://github.com/robottini/grbl-servo
M4 can't operate because pin 13 isn't connect to the servo (and can't !). 
You can modify evrything in the spindle_contol.c file (prescaller of the timer is modified)

Mat

DBSDBS, 2015-12-22:
Thank you very much Mat.

mat60, 2015-12-22:
Try to test it because it seems that it doesn't work (the servo moves but not doesn't do always the same motion with the same gcode) . For me the frequency of the pwm with a prescaler of 1024 is too high. The final frequency is 16Mhz/1024=15khz. To control a servo we need 1/20ms=50Hz.
Someone can help me/us ?

EliteEng, 2015-12-23:
@mat60 the formula for the PWM is
fPWM = fCPU / (PRESCALER * (1+TOP))
fPWM = 16,000,000 / (1024 * (1+255)) = 61Hz for the UNO
to get it down lower you can turn on phase correct PWM which will divide it by 2.

I am not exactly sure what you need to drive a servo, if you can let me know exactly what you are after I can let you know how to configure the timers.

DBSDBS, 2015-12-23:
@EliteEng,

I want to use the servo to drive a shutter for the DLP projector of my printer.  

It should turn around 120 deg because of limitations of my setup. 

Of course it is a good idea every time the shutter to go on the same position, not on a random ones.

Thanks to all for the ideas. I'll wait for more opinions / modifications.

mat60, 2015-12-23:
@EliteEng,

Thanks for your response. So the PWM frequency is good. 

The main problem I have is sloved. 
In fact it was an hardware problem. The servo was suplied with the usb of the arduino (the steppers drivers too). Now the arduino is powered by an  12V 2A external supply and it works. 

There is still a problem with the code. The SXX command must be between 15 (min) and 32 (max) and if you send 80 it works... 
I will try to modify it... 

NB : M5 works for me

AlexHolden, 2015-12-23:
@mat60 keep an eye on the temperature of the voltage regulator on the Arduino. It's a linear regulator, so with a Vin of 12V and a Vout of 5V it has to dissipate 58% of the input power as heat. If you use a regulated supply that puts out more like 7-8V, it will operate more efficiently.

mat60, 2015-12-23:
OK. 

Everything works with the spindle_contol.c file. Sorry for all the mistakes I do. 

You have to enter the min and max speed you will send in gcode. This is in config.h file :

#define SPINDLE_MAX_RPM 100.0 // Max spindle RPM. This value is equal to 100% duty cycle on the PWM.
#define SPINDLE_MIN_RPM 0.0    // Min spindle RPM. This value is equal to (1/256) duty cycle on the PWM.

In my case M3SXX, XX must be between 0 and 100. You can modify this value as you want (0 to 180 if you want to work in degrees...)
Then you parameter the hardware limits of your servo in the spindle_contol.c file. In my case
#define RC_SERVO_SHORT     6       // Timer ticks for 0.6ms pulse duration  (9 for 0.6ms)
#define RC_SERVO_LONG      34       // Timer ticks for 2.5 ms pulse duration  (39 for 2.5ms)     
In order to know the upper and lower limit, you can modify the end of the code line 109 (just to test it)
      #ifdef RC_SERVO_INVERT 
          //current_pwm = floor( RC_SERVO_LONG - rpm*(RC_SERVO_RANGE/SPINDLE_RPM_RANGE));
		  current_pwm = floor(rpm);
          OCR_REGISTER = current_pwm;
      #else
          //current_pwm = floor( rpm*(RC_SERVO_RANGE/SPINDLE_RPM_RANGE) + RC_SERVO_SHORT);
	      current_pwm = floor(rpm);
          OCR_REGISTER = current_pwm;

I hope it's clear.

DBSDBS, 2015-12-25:
Merry Christmas folks,

@mat60, 

So I made a serious number of tests, but the results are equal in fact.

I changed:

- in "spindle_control.c":

#define RC_SERVO_SHORT     15       - from 4 to 15
#define RC_SERVO_LONG      39         - from 32 to 39      
#define RC_SERVO_INVERT     1    - with // and uncomment

- in "config.h":

#define SPINDLE_MAX_RPM 100    - from 100 to 256

The maximum that I received was to find out how to drive the servo in 90 deg. diapason.
[image]

Initial state is on lower left.
The values lower than M3 S32 (and M5 too) drive the servo to the top.
Value of M3 S1000 drives the servo to horizontal left.

The values between M3 S50 and M3 S1000 moves the servo to some position - may be corresponding with thous values.

Every value > M3 S1000 do not moves the servo more than M3 S1000.

Closer values of S do nothing - for example if I already moved the servo to M3 S30, sending M3 S50 do nothing!

May be I do not understand anything from this, bu till now I had no problems with Arduino (Uno, Mega, RAMPS, etc.).

I'll try to work using thous results, but I'm still confused with this PWM.

Regards

mat60, 2015-12-28:
Are you sure to compile the good file ? Are you using GRBL as an exemple in arduino IDE ? 
If you doesn't modify the library and restart Arduino IDE, it doesn't work... 
I think your problem come from here... 
You can test it : make a big mistake into the spindle_control file like that. Line 26 you write
 #include "grbabc.h" (or what you want...) you save you restart arduino IDE...
Try to compile it to check. You must have a mistake. 

Tell me what's up...

DBSDBS, 2015-12-29:
I have 2 folders with files of GRBL:
- the folder that I unpacked the ZIP file;
- the folder in "C:\Users\......\Documents\Arduino\libraries", that appears after importing GRBL in libraries in Arduino IDE ;

I made the "Mistake" that you show me in spindle_control in the first  folder and tried to compile - no errors. Then I did the same in the second folder - again no errors.

May be I miss something

Thanks in advance

mat60, 2015-12-29:
Have you got two (or more) GRBL library like GRBL (classic) and GRBL Servo ? 
Search in your computer all the spindle-control file and find the good one...

DBSDBS, 2015-12-29:
@mat60

THANK YOU VERY MUCH !!!

I really had 2 * GRBL library's... I deleted the unnecessary ones and all goes correctly! 

Thanks again!!!

109JB, 2016-01-01:
It is good that you got it working. However I want to point out that your solution might not work in all cases. The reason is because an RC servo doesn't use the same type of PWM as is conventional. RC servos really use PPM or pulse position modulation. Many RC servos will work reasonably well on standard PWM as you have found, but there are some that won't. The reason is because RC PWM (really PPM) is supposed to have the same frame rate with only the high side pulse width varied. For example, an "on" signal of 1.5 msec commands a neutral servo position and 1 msec and 2 msec are the extreme positions for the servo. The "off" time varies to make the total frame rate equal to 50 Hz or 20 msec. As stated many servos are tolerant of standard PWM, but some, particularly older servos won't work correctly.

EliteEng, 2016-01-04:
Ok this is very easy to do with the Mega but hard to do with the UNO.
If anyone would like help to get a servo up and running on a Mega please let me know.

5shekel, 2016-02-02:
i just want to add that grbl-servo worked for us 
the issue @DBSDBS had (with the 2 grbl.h in the library) bit us as well. 
hope to see this mod getting into mainline grbl.

token47, 2016-09-19:
@chamnit what is the official state of servo support on grbl? I'm using grbl-servo from robotini's repo but it is based on 0.9i (we are already on 0.9j) and it seems a quick hack on spindle code. I'm thinking about porting it to the last version (maybe to trunk / master branch) and make it proper support with official configuration and own code (without touching spindle code). But i'm asking first to know if there is anything already being done right now and if you would accept the code for review (for entering the official tree) else.

Maybe use pins 9 or 10 (you could free a pin by joining some endstops), therefore using timer1. What do you think? I'm still reading the code to understand if this will cause any collateral effect. Timer1 is desireable because of the resolution to drive the servo.

Also, since the servo support could eventually work in parallel with all the other functions, I don't think it's the best option to hijack the spindle code as robotini did (because M03 and M05 could potentially being used for the spindle itself at the same time). There could be a different gcode specific to the servo. I know there is also some defined in the 3d printer world (it's M280 for setting angles and M340 for setting the microsseconds). And maybe some config option to optionally disable spindle and map the M03 and M05 operations to the servo at specific positions for those who cannot alter the software generating the gcode and want to make the servo be controlled by the spindle on/off actions. What do you think?

chamnit, 2016-09-19:
@token47 : Admittedly, I've been very selfish by hoarding whatever flash and memory I could get because of the memory and flash constraints of the 328p Arduino processor and the impending v1.0 coming out. When it completes, I'll be more open to accepting pull requests, especially at the Grbl-Mega repo, where there will be no such constraints on the Mega2560.

Timer1 is in use by the stepper algorithm on the 328p. So a no go there.

githubalbinolynx, 2017-02-01:
Dear Members,

please help me! 
I'm very newbie user, with rudimentary language skills. I would like to using the grbl servo, with usual peripherals: two stepper motor (X, Y axis), and one rc servo (Z axis = head moving).
I read this posts, but not clear the setup. My rc servo is a simple tower pro sg90 (180 degree).

How can configure the spindle_control.c file, for goow working? 
What is the connection between:
define RC_SERVO_SHORT 15 // Timer ticks for 0.6ms pulse duration (9 for 0.6ms)
define RC_SERVO_LONG 32 // Timer ticks for 2.5 ms pulse duration (39 for 2.5ms)

and the servo duty cycle?

What values can giving the M03 parameters?
My arduino board is uno r3.

Thank you!

lberstein, 2017-02-14:
@token47 Yes, please! Do add servo support to grbl 1.1! I'm in the same spot as many others here: I can get my setup (CoreXY Plotter) to work with robottini 0.9 version, but would really love to use 1.1.
As I have no need for spindle support I've tried to port robottinis quick hack of it into 1.1, but can't get it to compile (I have no knowledge of C, and spindle_control file has changed a lot from 0.9 to 1.1).

If you, or someone, could add servo control to GRBL 1.1 I'd be eternally grateful :)

MinSoeVertex, 2017-05-30:
Hi, I just tested cprezzi's grbl-servo and it works grate. It is grbl 1.1f and now I can connect with laser web 4. Hear is the download link https://github.com/cprezzi/grbl-servo.

jonnor, 2018-04-27:
The code from cprezzi looks pretty clean, with `#define` option to enable servo support. Would there be interest in a PR to get this support into mainline?

ardmed, 2018-06-12:
I just know how it work with my cnc and servo ,

ardmed, 2020-04-26:
> Ok this is very easy to do with the Mega but hard to do with the UNO.
> If anyone would like help to get a servo up and running on a Mega please let me know.

i have servo first time work will but now not working ? an u help 
....,,,..//// thnkks

shmuliko, 2020-11-11:
i fix this problem with simple code on second arduino board

https://github.com/shmuliko/stepper2servo

Faisalawan11, 2021-09-23:
> OK.
>
> Everything works with the spindle_contol.c file. Sorry for all the mistakes I do.
>
> You have to enter the min and max speed you will send in gcode. This is in config.h file :
>
> #define [...]

I am not getting any PWM output on Ramps 1.4 using Arduino Mega2560 and GRBL firmware,  i want to know from which output pin you got an output of PWM and did you change the code. Waiting your reply. Thank you

Faisalawan11, 2021-09-23:
> > Ok this is very easy to do with the Mega but hard to do with the UNO.
> > If anyone would like help to get a servo up and running on a Mega please let me know.
> 
> i have servo first time work will but now not working ? an u help
> ....,,,..//// thnkks

kindly tell how you run the servo on Ramps 1.4 Mega? what is the output pin and G code for GRBL to operate?