grbl/grbl #871: Configurable homing axis with $22 becoming a bitmask

BjoernKellermann, 2015-12-25:
Hi Dev Team,

I have a two axes compound table where I want to use homing on X and Y axis only.
Now I found out, grbl needs to be custom compiled to exclude Z axis from homing.

Could you change $22 homing cycle to a bit mask stating which axis are included.
0 : no homing (like today)
1 : homing on x, y, z axis (compatibility to today's behavior)
2 : only x
4 : only y
8 : only z
...
6 : would give my desired x and y homing.
14 : x, y, z as sum of 2 (x), 4(y) and 8(z)

This solutions enables homing for other two axis devices like laser plotter.
Would be great!

Happy X-Mas!

Bjoern

chamnit, 2016-01-11:
@BjoernKellermann : No. Compiling is very simple to do and compile-time options/macros save a lot of flash space. Grbl will begin to support a larger CPU soon and things like this will be addressed then.

closed 2016-01-11 by chamnit

cri-s, 2016-01-12:
as note, i don't know if this are relevant.
Bitmask should be converted on settings.c and reports.c
Same as $23  bit0=x bit1=y ... ., and note to myself , $L  display of
inputs state bitmap.