grbl/grbl #872: grbl on the Makeblock XY Plotter?

fogleman, 2015-12-25:
I just got a Makeblock XY plotter v2.0. It's pretty nice but the software and firmware both kinda suck.

I found grbl and it looks great. I already got grbl running on my device (I can interact over serial, e.g. $$ works) but the motors don't move at all no matter what I do.

I think it might be an issue with pin out. The Makeblock uses a customized board (Makeblock Orion) that's similar to Arduino Uno. Here's a pin diagram:

http://app.makeblock.cc/program/makeblock-orion/

I'm not sure if these numbers correspond to typical Arduino pins or not. For example, Port 2 on the Makeblock Orion shows pins 3 and 9. Port 6 shows A2 and A3, I don't know what this means in the Arduino world - e.g. how it might map to PORTD.

I found the cpu_map headers in the grbl source but I'm not sure if that's where my problem is.

Any ideas? Has anyone used grbl with the Makeblock successfully?

kajackdfw, 2015-12-30:
One option is just remove the controller , and plug in a regular Arduino with a grbl shield from Amazon for $26 . Search for  Arduino CNC kit / CNC Shield V3.0 + uno R3+4pcs a4988 driver / GRBL  .

cri-s, 2015-12-31:
From makerblock

int X_STEP_PIN = 8;
int X_DIR_PIN = 11;
int X_ENABLE_PIN = 4;
int X_MIN_PIN = A4;
int X_MAX_PIN = A5;

int Y_STEP_PIN = 12;
int Y_DIR_PIN = 13;
int Y_ENABLE_PIN = 4;
int Y_MIN_PIN = A1;
int Y_MAX_PIN = A0;

int Z_STEP_PIN = A3;
int Z_DIR_PIN = 9;
int Z_ENABLE_PIN = 4;
int Z_MIN_PIN = A1;
int Z_MAX_PIN = A0;

Now go to https://bigdanzblog.wordpress.com/2015/01/30/cant-get-i2c-to-work-on-an-arduino-nano-pinout-diagrams/
And check out arduino uno.
Pin 9 is PB1 and A1 is PC1
Pb1 means PORTB and bit 1

beardicus, 2016-02-10:
Did you ever make any headway with this @fogleman ? I've poked around a bit and I see potential issues:

1. Servo for the Z-axis. Various forks of grbl have supported this, but I don't think official grbl currently does. You may be able to take over the spindle speed PWM to do the job, but I'm not sure you could get it to be the right frequency.
2. Many functions (limit switches, stepper pulses) need to all be on the same AVR port. I can't be arsed to translate all the weird Makeblock hookups, but at a glance it looks likely some are split between ports.
3. Makeblock has an enable pin for X- and Y-axis, whereas grbl only uses a single enable pin.

There may be other issues but that's about enough for me to recommend you just get a $10 Arduino Uno-alike and use that. It looks like all the Makeblock electronics modules have a place to wire them up without using the RJ45 jacks. That just leaves the servo issue. I'm sure you could easily rig up a small solenoid in its stead.

fogleman, 2016-02-10:
@beardicus Yeah, that's a good summary of what I found (and more). I basically decided not to use grbl and I just tweaked the Makeblock firmware.

jonmccon, 2016-08-13:
@fogleman Same issue with the software, how were you able to tweak it?
There's a scaling / positioning preference in the previewer when I bring in svg artwork (even with consistent original sizing) that I've found any sort of registration consistency between prints to be awfully loose.

LordPueblo, 2018-03-28:
Hi Guys!
How did you solve the problem with grbl and makeblock ploter xy ? Maybe someone hase an INO file to upload to Arduino? 
I can see the program LaserGRBL is working but cannot move the motors. Is this problem becuse of incorrect ION file uploaded to Arduino?

Shreiya, 2018-04-20:
Hello!
Checking on this as well. I plan to try plugging in the motors tomorrow and was hoping to see a moving miracle.
Also, are there any alternatives other than the g-shield using GRBL or the makeblock board that anyone has successfully managed?

jonmccon, 2019-05-25:
@fogleman Bumping this one, Are you able to share the tweaked firmware? 
Thanks.

fogleman, 2019-05-26:
@jonmccon Sorry, I can't find it! I stopped using the Makeblock quite a while back.