grbl/grbl #874: Errors when flashing to Arduino Uno

AKEric, 2016-01-03:
Got a new x-carve, and it seems like something may be wrong with the Arduino that it shipped with.  I wanted to flash one of my other two Uno's with grbl, but they're both giving me the below errors from the Arduino IDE.  Note that if I reflash the Arduino that came with the x-carve, it works.  But doesn't fix the problem.  The problem, if you're wondering, is that the limit switches worked for one day, then sort of 'just stopped'.  Checking the voltage on the arduino pins for those switches, they always read 0v.  So I wanted to plug the gshield into another Arduino to see if it gives me the same results, but... no flash :(

avrdude: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x0d
avrdude: stk500_cmd(): programmer is out of sync
And sometimes this:
avrdude: stk500_recv(): programmer is not responding

The only difference is that the uno that came with the x-carve has the 'small chip' on it, while the other two I have both have the bigger caterpillars soldered to it.  The chipset is the same though, 328p.

Any thoughts?  Since I'm on mac, I also tried the hexuploader, but I couldn't get it to work for *any* of the arduinos, I'd usually gave me the bottom error.

Thanks

AKEric, 2016-01-03:
For what it's worth, here's what I get when I try to do it form the commandline:

/Applications/Development/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Development/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -pm328p -cstk500v1 -P/dev/cu.usbmodem411 -D -Uflash:w:inventables_grbl_09j_2015_07_17.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

------

So I try it with the -F, and get this:

/Applications/Development/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Development/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -pm328p -cstk500v1 -P/dev/cu.usbmodem411 -D -Uflash:w:inventables_grbl_09j_2015_07_17.hex -F

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
avrdude: Expected signature for ATmega328P is 1E 95 0F
avrdude: reading input file "inventables_grbl_09j_2015_07_17.hex"
avrdude: input file inventables_grbl_09j_2015_07_17.hex auto detected as Intel Hex
avrdude: writing flash (28054 bytes):

Writing | ################################################## | 100% 5.25s

avrdude: 28054 bytes of flash written
avrdude: verifying flash memory against inventables_grbl_09j_2015_07_17.hex:
avrdude: load data flash data from input file inventables_grbl_09j_2015_07_17.hex:
avrdude: input file inventables_grbl_09j_2015_07_17.hex auto detected as Intel Hex
avrdude: input file inventables_grbl_09j_2015_07_17.hex contains 28054 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 4.21s

avrdude: verifying ...
avrdude: 28054 bytes of flash verified

avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.

----

Which seems to do something, but I can't get grbl to respond when connecting over a serial connection.

kosme, 2016-01-03:
Without any more information, it seems like you are selecting the wrong port. Arduino ports should look like ttyUSB0 or similar.

chamnit, 2016-01-03:
@AKEric : This looks like a problem with your boot loader. Try reflagging th bootloader by using one of your Arduinos as an ISP. Google on how to wire this up and do this. 

As for limits not working, this is 99% due to electrical noise in your limit switch wiring. Noise cannot be picked up by a voltmeter. Only by an oscilloscope

AKEric, 2016-01-03:
@kosme : It's the right port, I can connect to it via the arduino IDE.  It's the same port that chilipeppr connects to as well on the 'working' ardiuno.

@chamnit : I was afraid of that.  I'll give that suggestion a shot next.  

Thanks!

closed 2016-01-11 by chamnit