grbl/grbl #875: 2 step outputs required rather than Step and Dir...

TomsGrbl, 2016-01-04:
I'm using an Arduino Uno, ATm386 (trying to recall from memory at the moment) and the grbl 0.8c hex file. It appears that this code is used with the easy driver from Sparkfun which requires a step signal and a direction signal. I'm attempting to use a BD7902FS that i'm salvaging from a CD-Rom drive :) and this chip seems to require input step signals for both "coils" rather than a step and dir signal. I connected my scope to the BD7 chip while still on the CD-rom PCB and found the signals to go "high" at about 1.65vdc and have an AC waveform ontop of the DC that fluctuates close to 1v pk-pk (closely resembles a sinewave) when the motor is moving and appears to be close to 60Hz. The signals at the two input pins are close to 180 degress out of phase which makes sense for the stepper motor control. I was thinking that I could remove the chip from the board and use it as a stepper motor driver rather than purchasing one. I'm making a diy 3D printer with 3 cd-rom drives. If anyone has any advice or info it would be greatly appreciated. Thank you.

chamnit, 2016-01-11:
@TomsGrbl : Unfortunately this is a very specific application and won't be something that Grbl will support. You'll have to modify Grbl's source code (see stepper.c) to do what you want it to.

closed 2016-01-11 by chamnit

X3msnake, 2016-01-13:
@TomsGrbl

You would be better off taking a different approach to the project, why
don't you develop a hardware driver that responds to the GRBL standart

you can use a pic or something to make a small circuit that drives the
Driver based on the GRBL standart input