grbl/grbl #877: Variable spindle speed

athulsnair, 2016-01-06:
Hi,
  I need someone's help in setting up variable spindle in GRBL. I like to use this feature for controlling laser output.
Thanks

asm7100, 2016-01-08:
How about M3 Sxxxx Where xxxx is a number.I use this to control my laser :-) CNC Shiel modyifyd to use 0.9j :-)

athulsnair, 2016-01-11:
@asm7100 
thanks for the reply.
I already fixed it, changed #define MA from 1000 to 255 so I could get max laser for s255 command.

Which software are you using to generate gcode, for cutting and engraving?
Any recommendations?

chamnit, 2016-01-11:
@athulsnair : Grbl will have a special laser mode in v1.0 so it doesn't have to pause between 'S' spindle speed changes. Out very soon.

closed 2016-01-11 by chamnit

athulsnair, 2016-01-13:
@chamnit I couldn't finga software that generate proper gcode for grbl to use with laser, do you know any?

 I tried to generate g code using standard generators for CNC, but if I run my laser with that code,  (say gcode for a circle and then a rectangle), laser will be on when machine moves from circle to rectangle and making unnecessary cuts since spindle need not to be turned OFF during tool paths.

schlotzz, 2016-01-13:
I'm using InkScape with the LaserEngraver plugin. A short description with links to InkScape and the plugin can be found on my github project page https://github.com/schlotzz/CoreBeam

asm7100, 2016-01-13:
@athulsnair Hi sorry for the late replay
I Use f-engrave-151 For text and in a combine with https://github.com/vlachoudis/bCNC A nice program :-)

prcdslnc13, 2016-01-14:
Hey Guys Im working on the same thing right now. Check out LaserWeb for your code generation and machine control software by Peter Van Der Walt : 
https://github.com/openhardwarecoza/LaserWeb

Are you controlling a diode or a Co2 laser?

athulsnair, 2016-01-16:
@prcdslnc13 I already tried laserWeb to run the machine, but I couldn't generate gcode for DXF files using laserweb(are you having the same issue?). Seems like it's having several other  issues. I did engraving using LaserWeb by uploading a B/W png image.

I'm using a 60W CO2 laser.