grbl/grbl #878: Tangential A Axis

jackjameshoward, 2016-01-07:
Does anyone know how hard it would be to a forth A axis for use with a tangential knife. 
I will not need the spindle anymore so that should free up the pins I guess.

cri-s, 2016-01-07:
https://github.com/openpnp/grbl

jackjameshoward, 2016-01-07:
I noticed that in the file nuts_bolts.h the following is commented out
// #define A_AXIS 3
Has there been some effort to incorporate an A axis before in the grbl branch.

chamnit, 2016-01-11:
@jackjameshoward : Grbl will support fourth axis in v1.0. Coming out very soon.

closed 2016-01-11 by chamnit

cri-s, 2016-01-11:
Why not more then 4 axis ?

jackjameshoward, 2016-01-12:
@chamnit thanks for the response. Any chance of a heads up if the pin configuration is changing? I'm just about to buy a shield

hilam8899, 2018-01-05:
The 4 axis  millI I have finished!Now I want to do  4th axis cutter.  how I do tangential for 4th axis?my 4th AXIS is C (banlance Z AXIS  ),Thank you!

109JB, 2018-04-24:
There are already some forks of Grbl that support 4 to 6 axes.

https://github.com/UncleRus/grbl-mega-clean-6-axis
https://github.com/LETARTARE/grblQ-Mega-4axes
https://github.com/dguerizec/grbl-Mega-4axis/tree/0046fa9d386b0c5fa219a0cede47b5c339ea9cc8
https://github.com/gcobos/grbl4axis (This one eliminates spindle and adds 4th axis E for 3D printing. Could work for OP's application)