grbl/grbl #879: Stuck with CoreXY setup

nccwarp9, 2016-01-07:
Hello, 

I tryng to build a corexy machine (without z). Went over the code and enabled ifdef COREXY and changed homing as described in commented sections of code. But when I try to run it is still works like Cartesian setup. Im using universal gcode sender for testing. 

any ideas what I missed to check ?

cri-s, 2016-01-07:
Probably you must enable homing on xy for homingcycle2.
If you da homing on z for first cycle.

nccwarp9, 2016-01-07:
config.h file is setup as

#define HOMING_CYCLE_0 (1<<X_AXIS) 
#define HOMING_CYCLE_1 (1<<Y_AXIS)

#define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable

still no joy.

cri-s, 2016-01-07:
try this:

#define HOMING_CYCLE_0 (1<<X_AXIS)|(1<<Y_AXIS)
//#define HOMING_CYCLE_1 (1<<Y_AXIS)

//#define HOMING_CYCLE_2 // OPTIONAL: Uncomment and add axes mask to enable

nccwarp9, 2016-01-07:
Works! Thanks !!

closed 2016-01-11 by chamnit