grbl/grbl #880: Negative dimensions.

trasz, 2016-01-08:
I'm a fairly bit new at all this, so I suspect I'm missing something very basic.

After completing the homing cycle ($H), the spindle stops at -1, -1, -1.  The problem is, trying to eg do "G0 X10" results in motion in the wrong direction - it goes right into limit switch, the one used for homing, instead of the other way.  Doing "G0 X-10" does what I wanted - but why do I need the minus?

Grbl version is 0.9j, config values at defaults, the only change is that I've disabled spindle PWM to make it work with the CNC shield.  What am I missing?  Thanks!

109JB, 2016-01-08:
I suggest that you read the LinuxCNC user manual relating to using G-code as this is very basic.

There are several coordinate systems and after homing the machine, coordinate system (G53) is set to -1,-1,-1.  There are 6 other coordinate systems that you can use (G54-G59), and you can set these to whatever you want to. After homing the machine and it being at G53 == -1,-1,-1, you can type G10 L20 P1 X15 and it will set the current position as x=15 for the G54 coordinate system. P1 sets G54, P2=G55, P3=G56...

As said, this is very basic and you would be better off reading about how G-code works because a book could be written about it, and books have been written about it.  You can find the LinuxCNC user manual at the link below. Not all of the G-codes in LinuxCNC work in GRBL, but the ones that do are in there.

http://linuxcnc.org/docs/2.6/pdf/LinuxCNC_User_Manual.pdf

closed 2016-01-09 by trasz

109JB, 2016-01-09:
The values in the "$#" response ARE NOT coordinates.  These are the offsets from the G53 coordinat system.

There is always one active work offset that corresponds to the G54-G59 work coordinates. The active work coordinate has it's values displayed in the WPos output from the GRBL "?" status query.  The MPOS output is the G53 coordinate system.  You can't change the G53 but can change the G54-G59 by using the G10 command.

trasz, 2016-01-10:
Okay.  I think this aligns with how I understand all this, but just to make sure: if the offsets are zero, as they are by default (ie with Arduino freshly flashed with GRBL), the coordinate system is supposed to be negative; and when you set up the offsets by "G10 P1 L20 X200 Y200 Z30", then you can use the normal, positive coordinates?

chamnit, 2016-01-10:
@trasz : Yes. That looks about right. You can also alter how Grbl sets its initial homing position in config.h, if you find the default method unsuitable. However, it's really all arbitrary. The config option HOMING_FORCE_ORIGIN will set 0,0,0 machine coordinates wherever the homing cycle completes.