grbl/grbl #882: Engrave mode for GRBL

Moffy, 2016-01-12:
Engraving Mode for GRBL

I have an idea for enabling GRBL to automatically engrave i.e. incorporate the function of autoleveller into GRBL.
It would start with an area probe being done controlled from a PC. The results would be stored on the PC and a bilinear interpolated grid would be created say with up to 1024 points. This grid would then be loaded into GRBL and used to calculate the zdelta based on the current x and y position. The zdelta would be calculated in the program loop and would give a close approximation of the offset. I have some code to illustrate:

#include "grbl.h"

//For ATMega644PA with 4k of RAM
#define ZOFFSET_MAX_PTS		1024

int16_t  zoffsets[ZOFFSET_MAX_PTS];	//Array of z offsets to material surface
int16_t  znew, zold, zdelta;		//Used to calculate the current offset
uint8_t  x2divisor, xpts;			
uint8_t  y2divisor, ypts;
bool	 engrave;					//Enable engrave mode

void zoffset_calc()
{
	int32_t x, y;
	uint16_t i;

	x = sys.position[X_AXIS];
	y = sys.position[Y_AXIS];

	x = x >> x2divisor;
	y = y >> y2divisor;

	i = x*ypts + y;
	if(i < ZOFFSET_MAX_PTS)
	{
		znew   = zoffsets[i];
		zdelta = zdelta + znew - zold;
		zold   = znew;
	}
}
//
void zoffset_load(uint16_t i, int16_t z)
{
	if(i >= ZOFFSET_MAX_PTS)
		return;

	zoffsets[i] = z;
}
//
uint16_t zoffset_checksum()
{
	uint16_t zchk=0, i;

	for(i = 0; i < ZOFFSET_MAX_PTS; i++)
	{
		zchk = zchk ^ zoffsets[i];
	}

	return zchk;
}

The zdelta would be used in the stepper section to add up or down pulses to the z axis.
A few new commands would need to be incorporated into the protocol, such as load the zoffset data but nothing very complicated. The engrave capability would be controlled by a single boolean variable "engrave".

What do you think the best way to implement it would be?

109JB, 2016-01-13:
I think the problem at this point is the available storage space on the current 328P boards. In order to do what you describe, GRBL would have to store the array of points to use in calculating the Z-delta.  Space that it doesn't have. It may be something that could be added once a new board is used/supported, but the auto-level feature is already implemented in several user interfaces already.

Sonny may have more insight into this and he is the final say so on new features.

asm7100, 2016-01-13:
How about using mega board in the further :-)

Moffy, 2016-01-13:
Yes, the current 328p could do it if say 1k was set aside for the array, that gives 512 points which is quite useable, and there are versions of GRBL that only use 1k(less serial buffer space). But aside from that even if you don't agree with the RAM requirements, how would you implement the stepper code for the zdelta?
The user interfaces are good but they have to hack the GCODE and can greatly increase its size, it would be nice for the controller to track the z axis, clean and simple, at least I hope so.

Moffy, 2016-01-28:
Since no one has responded to my question after some thought I'll answer it myself. The zdelta can be compensated for in the main stepper ISR.
1. Run it at a low rate so that planning is not necessary, say 500Hz. It just means you need a counter in the stepper ISR and every time it reaches a preset count you add a z pulse based upon zdelta.
2. If a z move is in progress then no pulses are added. This avoids high step rates.
3. Each time the main loop processes it calculates a new zdelta based upon the current x and y position.
Though the zdelta rate is low it should be more than enough to follow the contours of a basically flat surface such as a PCB. The first zprobe would be done after a short delay to allow the zdelta time to be compensated. It's a simple scheme but should work.

chamnit, 2016-01-28:
@Moffy : By doing it this way, you circumvent the acceleration planner, which can lead to lost steps. It's also not very robust. @109JB already stated that this type of feature is beyond what the AVR can do, possibly even the Mega, is CPU cycles are still required to make the calculations. 

Also this feature is fairly specific and isn't suited for a lot of different scenarios. It's still much better for a GUI to handle these things, as the g-code isn't that complicated to modify. Occam's Razor. The simplest solution is often the right one.

closed 2016-01-28 by chamnit

Moffy, 2016-01-28:
Thank you for your response but I of course disagree. The GUI gcode modifiers take a 2k file and can easily turn it into a 80k file, not the simplest solution. You are correct in that it is a specific feature, engraving, and perhaps not suited to the more generalised GRBL. It would not take many cycles as my initial code shows as it is only a look up table no calcs required. What it does require is some RAM 1-2k. 
The slow rate means it doesn't need acceleration planning and should be robust.

But thank you for all your fine work on the project.

chamnit, 2016-01-28:
I also disagree. It doesn't matter how big a g-code program becomes when it's been modified. The user would generally never see it and could be accomplished by something as simple as a script. It can also be a much higher resolution because there are infinite resources when compared to a micro controller. BTW, I've seen 3D machining tool paths that exceeded a couple million lines of g-code and Grbl executes the whole program without a problem.

Second, robust means there are no potential points or conditions for failure. Even though your suggestion will likely move slow, it doesn't mean that will always be the case. A user with an unexpected application still could be commanded to go faster than the limits of a machine or create situations to lose steps. Everything needs to go through the planner, such that, all motions are acceleration planned.

Don't get me wrong. I understand your argument that the controller can easily make these necessary calculations and adjust its motion in realtime. However, memory is still a critical requirement and 1-2K RAM may not be enough for all applications. Plus, there is the question of how the do the curve fitting between the probed points, as linear interpolation is not the only way. There are multiple other methods, and which one is best is usually problem specific.

Moffy, 2016-01-29:
Thank you for the reply. The application is very specific i.e. PCB engraving or flat plate engraving, you can safely choose a rate that will work for all planning modes. The main problem would be if you exceeded the maximum tracking rate of the Z axis, but this really isn't a problem as there are simple work arounds i.e. slow the feed rates for x and y. No interpolation necessary, as it assumes a moderately flat workspace, just spot corrections which would be very effective, say 5mm apart. The issue is the RAM which would require an AtMega644p or such with 4k RAM. If the flatness as detected by the probing program exceeds a maximum dz/(dxdy)(excuse the abbreviation) the user could be notified.
I understand that this is not a generalised solution but a specific solution that has considerable limitations, but it is workable in most PCB engraving situations.
Thanks again for your time.

burlizzi, 2021-04-22:
> Since no one has responded to my question after some thought I'll answer it myself. The zdelta can be compensated for in the main stepper ISR.
>
> 1. Run it at a low rate so that planning is not [...]

@Moffy I am starting now to work on a similar project, if you are still interested. so far I implemented some kind of "Marlin G29" solution, it does autonomusly the probing and stores in an array (gives output on serial too) it doe'snt use the data though. I found this issue looking for info about that.
@chamnit the solution to the acceleration planner could be to do the calculation in the planner itself, a couple of multiplication for a bilinear implementation won't kill the CPU.
about the use case scenario, recently engraving is probably the most common reason for this software, I would, at least create a branch for that.

Moffy, 2021-04-22:
I have written most of the code for the autotrack z, I am using a customised iMXRT1052 board based upon the SEEED ArchMix. The code is as yet untested and I have modified Candle to do the probing and communicate to the micro. I haven't uploaded much lately as I have been busy prototyping my own board, who knew pcb toner transfer could be so tricky.
I have my own meager site with updates at: https://github.com/Moffy/grblHAL/issues/1
Cheers.
Will upload progress soon.