grbl/grbl #883: Help with 2.5D milling machine

Louisdew, 2016-01-15:
I am trying to use grbl to control a 2.5d milling machine. My Z axis is a solenoid. I was trying to use the Z Dir pin to control it, but this doesn't work well. It goes down OK with a -Z command, but as soon as the next XY command is sent it goes back up. The sender still thinks it is negative however. 
Is there away setting that can be configured to make the Z direction pin ( or another pin) behave this way?
Thanks,

djowen40, 2016-04-25:
Anyone had any advances on this? I am doing exactly the same thing. 

My current plan is to connect a d-type flip-flop to the direction and step outputs for the z axis with direction as the data and step as the clock. Hence latching the direction state in each time a new step is taken. 

If anyone has a grbl firmware fix for this I would love to here it. It took a quick look in the code and it was quite obvious that it would take me a while to find the code responsible let alone make a change without breaking something.

maaadbob, 2016-04-25:
You should be able to achieve this by connecting DIR to data pin and STEP to clock pin, 
since the DIR output is set in advance to the STEP.
Make sure you use digital ICs that support your GRBL board's voltage.

Why do you need to do this anyway ?  What drivers are you connecting to and how is the 
electrical interface designed ?

On my Proxxon FF230 which is being converted to CNC, I am using Leadshine DM856 and I 
simply connect my DIR and STEP pins to the driver.
Since DIR and STEP are set to output high when active, they drive the LEDs in the DM856. 
The LED cathodes are simply connected to 0V.

Bob.

chamnit, 2016-04-25:
This isn't a problem with Grbl, but rather a misunderstanding of how its coordinate systems work (which is standard on subtractive CNCs).

closed 2016-04-25 by chamnit

langwadt, 2016-04-25:
it could work for something like a plotter, negative moves become pen-down, positive moves pen-up. 
I think a single line hack will do, something like this: 

`
uint8_t dirmask = (1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT) | (st.dir_outbits & (1<<Z_DIRECTION_BIT));
 DIRECTION_PORT = (DIRECTION_PORT & ~dirmask) | (st.dir_outbits & dirmask);
`

djowen40, 2016-04-26:
@maaadbob I am not sure if you response was directed at me or @Louisdew. if it was directed to me the hardware will be a custom build. based on a nano and DRV8825 drivers. The machine itself is an old PCB circuit Mill the original drivers are RS232 based and the software is long since defunct. 

@chamnit  Sorry if i offended i did not intend to imply it was a problem with grbl's design but more just something i was trying to achieve for my particular solution. grbl is a great open source project and i like many others appreciate all the hard work people have put in. 

@langwadt 2-d plotter is essentially the design concept. The Z axis height is set manually per run. Thanks for the advice i will have a poke around and post back as to the results. If it works I figure there may be other people interested. Although i agree that the applications are limited.

chamnit, 2016-04-26:
@djowen40 : My bad. I misread the question. I think I know where the problem is and it involved a relatively recent change on how the direction pins are set. I'll have to check to make sure that it resets itself from the last known direction. However, if your machine can tolerate stopping whenever the solenoid moves, you can use the coolant or spindle enable/ccw pins to control your solenoid, rather than the z-axis.

djowen40, 2016-04-28:
Hi All,

@langwadt was close to correct. Unless i am holding my face wrong. It would seem the the new variable (dirmask) needs to be volatile and the test for the if statement needs to test the exact condition for a step on the z axis. I tried but what @langwadt wrote and a logical not variant if(!(st.step_outbit ... but neither worked the if statement returns true permanently regardless of what the z axis is doing and hence traps the Z direction state permanently

What did work is as follows:
For those who cant find it the changes are made in stepper.c under the ISR(TIMER1_COMPA_vect) function. Note i have not tested the adverse effects on the length of the ISR runtime but i expect the effect to be minimal

Comment out the line:
`DIRECTION_PORT = (DIRECTION_PORT & ~DIRECTION_MASK) | (st.dir_outbits & DIRECTION_MASK);`

Then add the following two lines:
`volatile uint8_t dirmask = DIRECTION_MASK; 
  if((st.step_outbits & 1<<Z_STEP_BIT) != 1<<Z_STEP_BIT) dirmask &= ~(1<<Z_DIRECTION_BIT);`

The two new lines test if the X axis is to make a step. If it is then the DIRECTION_PORT is updated with the Z axis direction. if the z axis is not taking a step the the odd value of the Z direction is left untouched in the direction port. (in actual fact you are not capturing the odd state but simply not clearing it fro the current port state.

Hope this helps anyone trying to build a 2D plotter / 2.5D CNC which uses a solenoid activated Z-axis. 

PS just in case it is not perfectly obvious after the hack you can use the direction pin for the z axis as a on/off signal for your solenoid.