grbl/grbl #884: v-plotter VS grbl

ps1x, 2016-01-18:
Hello!

I built wall plotter with cncshield (GRBL), junk printer steppers and GT2 belt, however only problem now is software because my plotter have polar coordinates. What i tried so far was this topic in german thread of arduino forum: http://forum.arduino.cc/index.php?topic=226277.0 with nice solution, but using older GRBL (i guess something like 0.8). It works and converts coordinates on the fly, however when there is alot of points in g-code GRBL starts to slow down and even loose steps. Then i tried most recent GRBL firmware and it works flawlessly in terms of speed and alot of points. However i could not find out how to implement same functionality in code, because it changed ALOT.

Here i include part of planner.c file from german topic:
```
// V-Plotter calculations by 'jurs' for German Arduino Forum
// Details: http://forum.arduino.cc/index.php?topic=226277.0

#define XB 500.0
#define XA XB
#define YB 600.0
#define YA YB

void calc_ab(float x, float y, float &da, float &db)
{
  // db = b - b0
  db= sqrt((XA-x)*(XA-x) + (YA-y)*(YA-y)) - sqrt(YA*YA+XA*XA);
  // da = a - a0
  da= sqrt((XB+x)*(XB+x) + (YB-y)*(YB-y)) - sqrt(YB*YB+XB*XB);
}

void plan_buffer_line(float xx, float yy, float z, float feed_rate, uint8_t invert_feed_rate) 
{
  // Prepare to set up new block
  block_t *block = &block_buffer[block_buffer_head];
  float x,y;
  calc_ab(xx, yy, x, y);
  // Calculate target position in absolute steps
```
... 
Maybe some1 can point me where i can implement same function in newer version, then it will be just perfect.

Thanks for reading till that line :)

p.s. here you can see basic idea of wall plotter: http://www.patriciogonzalezvivo.com/2014/vPlotter/

chamnit, 2016-01-18:
@ps1x : I can't say for sure what will work, since I don't know what the old v-plotter code was doing or how the physics change. (Grbl assumes cartesian oriented axes.) But, it sounds like all they did was convert the polar command to a linear command in the planner, as you've shown. If that's the case, the alteration of the code should be similar. IMHO, it'd probably be easier to convert the g-code itself with a script and send the converted code to a stock Grbl.

ps1x, 2016-01-18:
Thanks for fast reply! I will try to create some sort of cartesian to polar g-code converter and post results here (if some :).

ps1x, 2016-01-18:
I just realised that there is no way to convert G02 G03 codes to polar coordinates, probably better way is to convert vector image to polar coordinates, then convert it to G-code...

closed 2016-01-21 by ps1x

Chicodarave, 2018-05-12:
Trying this in 2018... Cant find a solution :(

MMAwAD, 2025-03-21:
Hello, Trying the same any updates in this did any one succeeded in doing that ?