grbl/grbl #885: Touchy hard limits but no noise?

dgtlmoon, 2016-01-18:
Thanks for the great software!

One issue I have, my limit switches are on the negative axis, I have them wired to ground via a 500ohm resistor, Limit switch status reporting shows them all correctly as 000 (low) and I have the $5 limit invert set to TRUE, all is fine.

(Pin 9,10,11 connect to a single 500ohm resistor, the other end of the resistor then connects to ground, the +5VDC connects through the limit switches to the Arduino side of the resistor )

$23 = 7 so homing should happen in the negative direction to hit the limit switches.

However, when I goto move the gantry, it triggers a Hard Limit,  the status is still 000.. I don't see why it is triggering a Hard Limit.

```
$0	10	(step pulse, usec)
$1	25	(step idle delay, msec)
$2	0	(step port invert mask:00000000)
$3	0	(dir port invert mask:00000000)
$4	0	(step enable invert, bool)
$5	1	(limit pins invert, bool)
$6	0	(probe pin invert, bool)
$10	255	(status report mask:11111111)
$11	0.010	(junction deviation, mm)
$12	0.002	(arc tolerance, mm)
$13	0	(report inches, bool)
$20	0	(soft limits, bool)
$21	1	(hard limits, bool)
$22	1	(homing cycle, bool)
$23	7	(homing dir invert mask:00000001)
$24	200.000	(homing feed, mm/min)
$25	200.000	(homing seek, mm/min)
$26	10	(homing debounce, msec)
$27	5.000	(homing pull-off, mm)
$100	67.000	(x, step/mm)
$101	67.000	(y, step/mm)
$102	67.000	(z, step/mm)
$110	2000.000	(x max rate, mm/min)
$111	1500.000	(y max rate, mm/min)
$112	2000.000	(z max rate, mm/min)
$120	120.000	(x accel, mm/sec^2)
$121	50.000	(y accel, mm/sec^2)
$122	10.000	(z accel, mm/sec^2)
$130	2000.000	(x max travel, mm)
$131	2000.000	(y max travel, mm)
$132	2000.000	(z max travel, mm)
```

Here's the debug output
```
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:190.000,10.000,10.000,Buf:0,RX:0,Lim:000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:190.000,10.000,10.000,Buf:0,RX:0,Lim:000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:190.000,10.000,10.000,Buf:0,RX:0,Lim:000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:190.000,10.000,10.000,Buf:0,RX:0,Lim:000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:190.000,10.000,10.000,Buf:0,RX:0,Lim:000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:190.000,10.000,10.000,Buf:0,RX:0,Lim:000>
>>> G20 G91 G0  Y1
ok
ALARM: Hard limit
[Reset to continue]
```

If i disable hard limits, I can jog the machine and the Limit switches are all 000, no noise there

```
[verbose]<Run,MPos:0.000,6.940,0.000,WPos:190.000,16.940,10.000,Buf:1,RX:0,Lim:000>
>>> G91 G0  Y1
ok
[verbose]<Run,MPos:0.000,7.910,0.000,WPos:190.000,17.910,10.000,Buf:1,RX:0,Lim:000>
```

Much to peoples horror, I am using my GRBL on a https://www.stepcraft-systems.com/en/cnc-3d-systems/stepcraft-2-600

chamnit, 2016-01-18:
@dgtlmoon : Electrical noise would show up on an oscilloscope, not on Grbl's status reports (unless you're really lucky). Try using a small capacitor, wired in parallel with ground and the limit pins. This should help eliminate most of the noise. Also consider that ground/neutral can spike, causing a false trigger as well. So re-check your wiring.

Not sure why you'd consider it a horror, but the step craft CNC looks quite nice! :)

dgtlmoon, 2016-01-18:
@chamnit I can drag out the scope, but I just added a 10uF across them as you recommend, but still the same result. Why would it think the limit switch is hit immediately when $26 is set to 1000? I see no movement at all.

chamnit, 2016-01-18:
@dgtlmoon : Not sure why you have a 500 ohm resistor wired in the way you do. The simplest wiring is limit pin to switch to ground for both NC and NO switches. There are internal pull-up resistors in all of the limit pins. Also make sure your that the z-limit is on pin 12,not pin 11,  if you are using variable spindle (enabled by default since v0.9).

dgtlmoon, 2016-01-18:
@chemnit 500 ohm resistor is there to stop the over-current from the +ve line otherwise you're passing that 5V straight into the arduino, i've already burnt one arduino's input lines this way without a resistor (it was permanently stuck high on pin 9), the 500 ohm "shouldnt" make a difference. I am running this setup with the pins wired low by default and raised high by the NO switches closing the circuit.

To try your theory I wired the 9,10,11 directly to ground but i still get the same issue.

unsure :(

chamnit, 2016-01-18:
@dgtlmoon : Ok. I'd first disconnect everything from your Arduino and connect only the limit switches. Hopefully, this will rule out any influence on other parts of your system. If it's still behaving weird, it's your wiring, Grbl, or your Arduino. Also try using the Grbl's default '$' settings values to help determine if it's a weird settings bug.

dgtlmoon, 2016-01-18:
@chamnit thanks again, yeah I'm confident it's a wiring issue too

closed 2016-03-04 by chamnit