grbl/grbl #887: grbl 0.9j homing issue

smikey, 2016-01-23:
At first: Thanks for grbl, it works very well for me exept one thing:

I have a x-y-z machine but limit switches only on x and y. I want grbl homing on these axes but there is one problem:

i changed the code in config.h to
#define HOMING_CYCLE_0 (1<<Y_AXIS)  
#define HOMING_CYCLE_1 (1<<X_AXIS)

so at first the y axis is homed and then the x axis. y works very well, it drives back until it hits the switch and goes forward a little bit. after this the x axis should be homed but the only thing happening is the x axis moves forward a couple of steps two times. no backward moving and searching for the limit switch.

I don't think it's a wiring problem because the same happens if I switch the wires from X and Y. Then X works and Y doesn't.

Do you have any idea what to do?

jahnj0584, 2016-03-02:
has to be wiring. Do you have them both set to NO or NC?

chamnit, 2016-03-02:
@smikey : Yes. It has to be wiring or the switch. Please let us know if you resolve this issue, so I can close it.

smikey, 2016-03-23:
Sorry for the late answer! I thought it could be the wiring or the switch, too. 

But if I check it with the multimeter both switches seem to be ok. The switches are hall sensors which connect to GND if they are hit.

I'll get new cables and test it again.

auditofi, 2016-05-27:
@smikey could u tell me how u connect x & y switches to the board?? what pin and how.
thanks

smikey, 2016-06-08:
Obviously I managed to get this thing to work. I use a little PCB with an optocoupler between the Arduino and the mill. I think I had some problems with a proper GND-connection between the board and the switches. I'm doing some tests now.

chamnit, 2016-06-08:
@smikey : Thanks for letting us know.

closed 2016-06-08 by chamnit