grbl/grbl #888: soft limits on 0.9j

johnqube, 2016-01-24:
Hello

I'am have a hard time to figure out the soft limits behavior
My machine and work position are 0
When I set my soft limits with correct max travels, All positive axis valid moves are alarmed, only negative moves are working, (and breaking the machine going minus 0)
What I would expect would be limits working from 0 to +maxtravel
Do I miss something or is there a problem somewhere with grbl?

might it be related to my dir invert mask conflicting with soft limits?

$$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=800.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=151.000 (x max travel, mm)
$131=101.000 (y max travel, mm)
$132=31.000 (z max travel, mm)

 thank you for the great work you guys!

chamnit, 2016-01-25:
Machine coordinates are all in negative space after homing. Please use the work coordinate offset to set your origin to what you want. See g54-59 and g10L2 and g10L20 commands on how to do this.

closed 2016-01-25 by chamnit

johnqube, 2016-01-26:
First, as I don't have limits switches, my machine is not homed. Homing hits physical limits and breaks steppers (even with soft limits enabled). 
At startup with machine and work set to 0.0.0 only negative moves are allowed. Setting any work offset (G10 P0 L20 or L2 pos or neg) doesn't allow me positive machine moves. limits are working but in negative space. G54 did not help

chamnit, 2016-01-26:
@johnqube : Soft limits only work when you have a reference point on the machine. So you will have to have homing enabled and limit switches to use soft limits. Grbl doesn't do a lot of settings checks to prevent users from setting up their machines incorrectly. This has to do with the very limited flash space on the Arduino 328p. However, Grbl should have a check for enabling soft limits only when homing is enabled. 

I'm not quite sure I understand what you mean by only negative moves are allowed. But what it sounds like, you somehow enabled soft limits without homing and Grbl is assuming that your start location is the machine origin. Grbl homes at the back, right corner of the machine, such that the machine moves in all negative space. You use G10 to move the work coordinate origin to somewhere in the machine space.

johnqube, 2016-01-26:
Thank you for this enlightment. First my mistake I did not understood that soft limits needed hard switches, I believed by reading the doc it was a software workaround for switchless machines.
and yes GRBL has some warning when you set up soft limits without homing, smart!

I suggest the clarification of "soft limit" a bit misleading in the documentation
Maybe some kind of pure software homing and limits would be handy for switchless machines
I thank very much you for your time

AlexeyGusev, 2017-12-25:
Hi,

> Grbl  homes at the back, right corner of the machine, such that the machine moves in all negative space.

is there any possibility to configure grbl (0.9j) soft limits to work painlessly with homing into front left corner? In fact, `$23 (homing dir invert mask)` allows me to place limit switches wherever I want, and spindle being in the leftmost position with table retracted all the way back is the most convenient machine home for me. This results in X and Y axes moving in positive space only and only Z axis in negative.

I don't have limit switches on opposite sides of both axes, that's why I want soft limits, but grbl starts barking as long as machine coordinates are positive and `$20 (soft limits) = 1`. So I either need soft limits to work in positive space or configure X and Y axes to somehow work in negative space...right?

Any ideas? Thanks.

jahnj0584, 2017-12-25:
Mine homes in the far left with proper softimites. It's a matter of
applying the proper mask code for it to work.

AlexeyGusev, 2017-12-25:
> Mine homes in the far left with proper softimites. It's a matter of applying the proper mask code for it to work.

which mask code allows homing into left to work with soft limits?

AlexeyGusev, 2017-12-25:
I figured out the problem. You need to do homing routine _after_ MCU powercycle, this sets coordinates properly to negative value of (max axis travel value) - (homing pull-off).