grbl/grbl #896: spi possible for brushless spindle???

valpoder, 2016-02-01:
hello:

Several Brushless chips can be driven by PWM signal, but need to have the registers configured by spi....
i wonder if it would be possible to have an SPI interface and I/O pins . ?

Cheers !!!

alfredanil, 2016-02-01:
+1 from me, stepper motor drivers with integrated motion controller from Trinamics also use SPI interface for advanced control

valpoder, 2016-02-01:
Using standar spi:
i think there is a little of a Pin conflict :
Pin 10 CS  -- limit Y axis   ** maybe necesary
Pin 11 MOSI -- variable spindle pwm  **
Pin12 MISO -- limit Z axis     : not strictly necesary  
pin 13 SCK -- spindle direction    ***

http://www.gammon.com.au/images/Arudiuno_SPI_pins.png
https://cadduino.files.wordpress.com/2013/10/arduinogbrl.jpg

most important are MOSI (pins 11) and SCK (13)... maybe pin 10 can be done using a free pin..

an alternative solution would be using any free pin , and use a software bitbang SPI...

chamnit, 2016-02-01:
You'd have to move of the pins around and find the available flash space on the Arduino Uno to do it. It'd be easier to just use the Mega build of Grbl or port to an ARM for some thing like this.

valpoder, 2016-02-01:
i think there is no need for much flash:its only 7 register (16 bit )... 
if sent by bitbang , it would be rather small...

i agree about the pins... but if done by bitbang , no need to move anything...it could be done with TX RX & A4 A5 pins....this pins are free  

Cheers....
-

chamnit, 2016-02-01:
@valpoder : You're probably right, but I'm already trying to save a few bytes when I can. Grbl v1.0 will have no room left for just about anything after I'm done with it. In fact, some things may have to be disabled to get things to fit on an Arduino Uno. I'm hoping that won't happen though.

valpoder, 2016-02-01:
i see !!  thanks for the feedback ...  i think it would be great to add this functionality in the mega version.... GRBL would be just perfect if it could handle this type of motors...

109JB, 2016-02-02:
I may have an idea for a workaround for you guys that only costs as much as the cheapest arduino clone.

What if you left the arduino with GRBL alone and use a second one that simply reads the PWM from GRBL, and just outputs it again. You could use any pins you like and there would be no worry of how much space was used.

I know.  Kind of a hack way to do it, but would be simple and not require any changes in GRBL or the arduino with GRBL on it.

valpoder, 2016-02-02:
Personaly , i think that grbl feels some sort of incomplete if it cant run dc motor spindles...

I Agree 109Jb , it could be done with a cheap arduino, or one of those small 8 pin microcontrollers...a small board is necesary for the driver anyway.... A german guy made this already...

http://www.ebay.co.uk/itm/A4960-BLDC-Device-Board-Allegro-Speed-Controller-Arduino-Shield-Raspberry-Pi-/141814056268

The board is an arduino shield, but there is a version with an embeded atmega....all the code is there...

I still think it would be interesting to develop the small soft spi ( bitbang) within the grbl code using the free io...  If not i probably take this to smothieboard ...but it would be a shame.....grbl is just perfect

langwadt, 2016-02-02:
@valpoder you should be able to use a regular RC ESC to  run a brushless motor, control it with the spindle PWM set to produce the usual servo control waveform.

MrTrick, 2016-02-06:
I have a similar requirement so I'll comment here rather than in a new issue.

I want to control a head that's not a spindle, but several high-current discrete outputs. I'm aware that I will need a daughter-board or a slave module for this. (Imagine a printer head)
It needs to sync with motion, so must still go through grbl somehow. I'm willing to (try to) modify grbl to make this happen.

My questions are;

 1. If the head control commands are sent through G-code, what's the most suitable command type? Should I hijack "set spindle"? 
 2. How best to connect grbl board to slave module? If it could be a standard like serial or i2c or spi that's easier because I can just buy a module off-the-shelf, but CPU utilisation of grbl is probably critical already? (How many cycles can it take to "set spindle", before it interferes with the rest of the control system?)