grbl/grbl #898: Erronus feed rate

lalo-uy, 2016-02-01:
I am running 0.9j  with this settings
$0=10 (step pulse, usec)                                                     
$1=0 (step idle delay, msec)                                                 
$2=0 (step port invert mask:00000000)                                        
$3=0 (dir port invert mask:00000000)    
$4=0 (step enable invert, bool)         
$5=0 (limit pins invert, bool)          
$6=0 (probe pin invert, bool)           
$10=3 (status report mask:00000011)     
$11=0.010 (junction deviation, mm)      
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=254 (homing debounce, msec)
$27=3.000 (homing pull-off, mm)
$100=23.325 (x, step/mm)
$101=1351.350 (y, step/mm)
$102=790.000 (z, step/mm)
$110=1000.000 (x max rate, mm/min)
$111=300.000 (y max rate, mm/min)
$112=240.000 (z max rate, mm/min)
$120=5.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=240.000 (x max travel, mm)
$131=160.000 (y max travel, mm)
$132=120.000 (z max travel, mm)

I can move ok from x=0 to x=200 and calibrqation is perfect.
but if I send 
G0X0
G1X80F10
 it takes 1 minune to travel. aprox 8 times faster.
G1 movements in Y or Z are ok.

Any clue?

Thanks in advance.

Lalo

splitn2, 2016-02-01:
$100=23.325 (x, step/mm)
$101=1351.350 (y, step/mm)
$102=790.000 (z, step/mm)

you have quite a big difference in your ratio;s, your X step/mm is very coarse?

lalo-uy, 2016-02-02:
Yes, Y and Z have gears but X y. Direct drive belt.