grbl/grbl #900: Arduino Uno + CNC Shield v3

ErrorSnaKe, 2016-02-02:
Hello,
I have a problem with homing, hard/soft limits. I have Proxxon MF70/CNC-ready I successfully connect three stepper motors. On each axis I have a button (zero point) what I connected to X+, Y+, Z+. I tried different settings ($5, $20, $21, $22 etc) but it was fail. 

Can anybody help me? Where I did a mistake?

I want to move each axis to zero point every time where arduino power up or where I press $H button via Universal Gcode Sender.

Thank in advance

lalo-uy, 2016-02-02:
You have to check what kind of buttons do you have. Are them Normal Open?
The other end is tied to ground? In that case and with $5=0 you shuld get
+5V in the switch and 0V pressing it.

After that tray $H

Protoneer, 2016-02-02:
The V3.00 is not compatible with PWM Spindle enabled(GRBL v0.9). Turn PWM
of in config.h or use GRBL 0.8

ErrorSnaKe, 2016-02-03:
Hello,

According to this picture 
[image]

On the right side I have a block END STOPS and button for example axis Z+ I connect here? Correct?

On this two pin I have 0V, but on each pin and +5V - I get +5V 

I think I made a mistake during connect buttons can you explaine to which pins I must connect my buttons according to picture above?

lalo-uy, 2016-02-03:
In grbl version 0.9j the default is to have variable Spindle Speed. In
order to archive this SpnEn and Z limit pins has been swaped.

So you may need to conect the Z limit swtich in the SpnEn pin.

Protoneer, 2016-02-04:
Hi @ErrorSnaKe , this is an hardware issue and not really related to GRBL. For that reason I have invested some time and created an open for all Forum where you can ask questions and talk about the CNC Shield. 

[CNC Shield Forum Link ](http://forum.protoneer.co.nz/viewforum.php?f=6)

Please post your questions there and lets get it sorted for you.

@chamnit , are you okay with this?

brucect, 2017-08-10:
Hi, I have the Arduino R3 Uno controller and the mating CNC Shield. I am testing my DIY Laser Engraver. I have the laser control board plugged in for 12v power and the plug that goes to the board, but where do I plug it into?
Thanks for the assistance

https://cloud.githubusercontent.com/assets/10923482/12784160/249082ec-ca8e-11e5-8d2e-21b621a740de.jpg

lalo-uy, 2017-08-11:
> ca8e-11e5-8d2e-21b621a740de.jpg

It should go to the variable spindle pwm  pin D11 in arduino. The name in
the sheeld may be diferent