grbl/grbl #901: Question about Max Rates

timryder, 2016-02-02:
I'm trying to do a side project which will allow me to scale the Max Rates of an axis based off a custom M Code I created.  I'm logically calling the settings_store_global_setting function from another with arguments to dynamically change machine parameters. It seems to be working except every now and then when the value gets too large it appears to blow up the eeprom.  That is, the last setting which gets set is my Z-Axis Max Rate and then the remainder of the parameters are blank when I do a $$ in the terminal. If I do a reset back to defaults everything appears to be fine again. 

So would it be likely that i'm trying to set a value which is too large for the eeprom to store?

currently $112 (z max rate)  = 214748.3648. 

Which begs my question, what are the absolute maximum settings for Max_Rate and Max_Accel based off a atmega328p running at 16MHz? 

Thanks in advance

chamnit, 2016-02-02:
@timryder : This problem more has to do with the printFloat() function in Grbl. When I wrote it, all valid floating point values that Grbl would expect would fall below 10 characters in length, including a '-' and the decimal. This includes what I thought would be physically realizable. Your 3.5 meters/sec (215000 mm/min) is extremely fast.

For a settings value, the default is 3 decimal places to print. So the max valid number would be '999999.999' that it could print. Since you have 4 decimal places, this cuts this down to '99999.9999'. You're above this cutoff.

However, I plan to make this printFloat failure be more indicative, since it has come up once before when a user set a max travel value accidentally to 1000 meters.

timryder, 2016-02-02:
So then my initial thoughts were that the eeprom is all stored as a bunch of bytes so why does it really matter what the max setting is as long as it can be held within a standard def. IEEE Float correct?  

Which that would be something in the magnitude of ~ 4Mil?

chamnit, 2016-02-02:
@timryder : It's not the floating point value thats the problem, but the printFloat function that converts it to a readable string. To save flash, I completely re-wrote the printFloat function, but I had to make some assumptions and remove some standard error checking. I think it was on the order of hundreds of bytes.

closed 2016-02-03 by timryder