grbl/grbl #902: Driver unipolar motor

felipenegriO, 2016-02-06:
So, i'm Brazilian. Sorry if I dont

felipenegriO, 2016-02-06:
So, i have the unipolar motor and not have driver for the motor. 
I would like: I can change the dir and speep for 1,2,3,4.. In the steep.c code? You could help me? 
I make the driver motor buts I don't have sucess.
is this motor:
https://www.google.com.br/search?q=nema+23+unipolar&espv=2&biw=1242&bih=585&source=lnms&tbm=isch&sa=X&ved=0ahUKEwiBnYWQg-LKAhWJQ5AKHYidAkIQ_AUIBigB#imgrc=XNdic3B_tDD76M%3A

lalo-uy, 2016-02-06:
Grbl do not drive the movimiento estudiantil tor, only sets direction and
send the pulses.
You need a motor driver module, like the pololu are for the bipolar motors.
Or you can build a drive with a small microchip and power transistors.

El Friday, February 5, 2016, Felipe Negri de Oliveira <
notifications@github.com

kmaxo, 2016-02-09:
Opa Felipe,
You need a driver...
TB6560 ... ou 
A4988 Polulu
ou 8825
Search on Mercado LIvre do Brazil...

lalo-uy, 2016-02-09:
All these are driver for bipolar motors. Unipolar motor driver are not
very polular.

lalo-uy, 2016-02-09:
Check this http://www.ti.com/lit/ds/symlink/drv8805.pdf

X3msnake, 2016-02-10:
Also

here is a article of texas instruments of using a bipolar driver to control
a unipolar motor

http://www.ti.com/lit/ug/tidubb3/tidubb3.pdf

closed 2016-03-04 by chamnit

costy35, 2016-05-13:
Unipolar motors is not very popular but have a very cheap price! For this i modified grbl and now working for me with unipolar steppers.  See here: https://sites.google.com/site/costycnc/grbl_modified
All function is inactive because i use 8  pins for stepper and all other pins is disabled. But for my expectative is good!