grbl/grbl #904: GRBL repo seems to be missing release tags?

justinclift, 2016-02-10:
Someone on the Shapeoko forum mentioned they're having trouble getting the source for an older GRBL release.

Went looking through the GRBL repo... and it seems like it doesn't have tags defined for the different releases (eg 0.9g, etc).

Is that something that could be incorporated into the release process from here on? (and maybe historically, if the commit info is known) :smile:

chamnit, 2016-02-10:
@justinclift : This has been suggested before, but there are no official releases within each major version, which are separated by branches. v0.9 is always being patched or minor additions being added. It's easy to view the branch history and download the source code from the past. I don't find much value in tagging everything.

closed 2016-02-10 by chamnit

justinclift, 2016-02-10:
k, was a thought. :wink: