grbl/grbl #905: Speed Control in Non-Linear

rwensley, 2016-02-11:
Greetings,

I am using the PWM output to control a Universal Motor. These motors are highly non-linear in their speed vs applied voltage curve.  Using a manual potentiometer I reordered the following;
V     PWM     RPMactual     
0        0          5000
0.5    25.5      6100
1       51         7710
1.5     76.5     10144
2       102       11205
2.5    127.5    12533
3       153        14175
3.5     178.5    16400
4        204       19400
4.5     229.5     23570
5        255        30000

Using S codes here are the results;
S value        RPM measured
5000          5000
6000          5565
7000          6160
8000          6960
9000          7920
10000        9200
11000        10240
12000        10765
13000        11350
14000        12050
15000        12800
16000        13630
17000        14570
18000        15660
19000        16940
20000        18550
21000        20360
22000        22600
23000        25860
24000        28860
25000        30000
26000        30000
27000        30000
28000        30000
29000        30000
30000        30000

My SPINDLE_MAX_RPM = 30000 and SPINDLE_MIN_RPM = 5000

In spindle_control.c I entered this line

	  rpm = (-0.000000001 * pow(rpm,3)) + (0.00003 * pow(rpm,2)) + (1.1873 * rpm) - 911.79;

at 

      #define SPINDLE_RPM_RANGE (SPINDLE_MAX_RPM-SPINDLE_MIN_RPM)
      if ( rpm < SPINDLE_MIN_RPM ) { rpm = 0; } 
      else { 
        rpm -= SPINDLE_MIN_RPM; 
        if ( rpm > SPINDLE_RPM_RANGE ) { rpm = SPINDLE_RPM_RANGE; } // Prevent integer overflow
      }
	  rpm = (-0.000000001 * pow(rpm,3)) + (0.00003 * pow(rpm,2)) + (1.1873 * rpm) - 911.79;
      current_pwm = floor( rpm*(PWM_MAX_VALUE/SPINDLE_RPM_RANGE) + 0.5);
      #ifdef MINIMUM_SPINDLE_PWM
        if (current_pwm < MINIMUM_SPINDLE_PWM) { current_pwm = MINIMUM_SPINDLE_PWM; }
      #endif
      OCR_REGISTER = current_pwm; // Set PWM pin output

It was my hope that this would linearize the outout so that when S 10000 is entered I would get an actual rpm of 10000 in stead of 9200 as I do now.

However; this did not work out well.  FOr some reason making this change caused an S5000 to end up as 13000.

I realize that every spindle will be different so some form of a data entry will be needed to correct.

Could this idea be incorporated into a future release?

Thanks,
Ron Wensley

chamnit, 2016-02-11:
@rwensley : No, it's unlikely something like this will be supported. Here's why. Spindle rpms are often nonlinear, and under load, they rotate at a completely different rate than not under load. This is also dependent on the material you are cutting, how you are cutting it, and the material removal rate. It's very hard to accurately predict true spindle RPM without a closed-loop feedback system.

This is what spindle controllers are for. They take a signal or command to turn on the spindle at a specified rate. The spindle controller will read the running rpm and adjusts the power to the spindle to ensure it maintains the programmed spindle speed.

IMHO, it's better to have Grbl only output a signal that determines the spindle speed for a spindle controller, which is what you're wanting.

closed 2016-02-11 by chamnit

rwensley, 2016-02-11:
I failed to mention in my original post that I am using a closed - loop spindle controller.  The S command is the set point to the controller via PWM.  S10000 should set the spindle RPM to 10000 and the spindle controller will maintain that set point as the load on the spindle changes.

However; When I issue an S command the controller does not go to the correct speed.  

Since I am able to inject a DC voltage of 1.5V I would expect to get 10000 RPM and I do.  If however I use the external signal, the pwm from grbl, and set S10000 I do not get 10000 rpm instead I get 9200 RPM.  

It could be that the PWM to voltage conversion in the spindle controller is not accurate.  I can not change that as it is hardware in the controller.  

I should however be able to make a correction curve that will allow for the corrected rpm to be sent to the controller.

The gcode reads S10000 the speed_control.c routine interpolates this and send 10800 to the controller.  Now the user is asking for 10000 and the controller is producing and maintaining 10000.

You are correct I only want grbl to send out a signal that determines spindle speed but we live in a non-linear world and it is far from perfect.  Thus the need for the curve.

Ron

chamnit, 2016-02-11:
@rwensley : Agreed. But creating a nonlinear output is highly specific to a setup. Besides, it's relatively simple to alter the spindle control source code as you have already shown. It's not all that complicated. I will leave it as is.