grbl/grbl #908: hard limits trigger when i input g code after a successful homing sequence

martin-riddell, 2016-02-18:
Hi community i love the work that has been done with GRBL and i have found it mostly self explanatory how ever i am experiencing a problem when i try to input g code after a perfect homing sequence.
As soon as i input a G command GRBL controller replies with a hard limit alarm I have followed all of the wiring instruction including using screened cable and smoothing caps with pull up resistors were needed any help would be much appreciated.

chamnit, 2016-02-18:
@martin-riddell : Getting hard limits to work is very common problem, but it's not particularly due to Grbl itself. The majority of the time, it's bad grounding practices, inadequate shielding and EMI isolation, environmental causes (like something else powered near by creating EMI), or an incompatible circuit wired into the limit pin.

As I always tell Grbl users, disconnect everything from the Arduino. Simplify the problem. Add one thing at a time to determine what the cause is. It'll take a little time, but you usually find the cause immediately.

Mhawley1, 2016-02-21:
Using hard limits and homing at the same time was really glitchy. In the end I just gave up and used the regular homing sequence and then made use of the soft limit parameters. Saved me allot of headaches.

closed 2016-03-04 by chamnit