grbl/grbl #910: [Check Door] message received for no reason??

mariocaptain, 2016-02-19:
Dear team,
I am trying to port the stream.py program to C# (to be freeware). Things seem to be straight forward, but I actually found me scratching my head for more than a day now.
The problem are that:
1. I always receive a "[Check Door]" message after a couple of "ok" 's received, and things are put on hold. I have no clue how to disable this feature because I can't see any pins for a safety door on grbl. After a lot of searching I am at my wits end.

2.I found that for very short CNC program (total length < 127 including newlines & carriage returns) my PC (core i3) runs so fast that this line will never get entered into:
`while sum(c_line) >= RX_BUFFER_SIZE-1 | s.inWaiting() :`

It means that the computer is so fast that after sending all the blocks (total length<RX_BUFFER_SIZE-1) but still not receive any "ok" or "error:" from GRBL so it exits. Could you confirm this is something that should be fixed, or my bad?

I am using v0.9j.
Many thanks in advance!
Dave Nguyen

P.S. My C# code just in case
```
foreach (DataGridViewRow dataGridRow in dataGridView3.Rows)
{
	block = dataGridRow.Cells[0].Value.ToString();
	rxBuffer.Add(block.Length + 1); 
	while (rxBuffer.Sum() >= RX_BUFFER_SIZE || serialPort1.BytesToRead > 0) //RX_BUFFER_SIZE = 127
	{
		//wait for GRBL response
		string grblResponse = serialPort1.ReadLine();	
		if (!(grblResponse.Contains("ok") || grblResponse.Contains("error"))) //doesn't have "ok" or "error"
		{
			//OUTPUT SYSTEM MESSAGE
			if (grblResponse.Contains("<")) // Real time status report
				updateCoordinates(grblResponse); 
			//TODO: OTHER TYPES OF MESSAGES HERE
		}
		else //contains "ok" or "error"
		{
			dataGridView3.Rows[responseCount].Cells[1].Value = grblResponse;
			dataGridView3.Refresh();
			responseCount++;
			rxBuffer.RemoveAt(0); //remove the responded block from the FIFO
		}
	}
	serialPort1.Write(block + "\n"); //send block to GRBL                                               
}

```

mariocaptain, 2016-02-19:
After a lot of debugging, I found that the second block of these two blocks causes the problem:
```
(Code generated by My Program Name)
(@Feb/19/2016 16:26:20)
```

The first block receives an "ok" after sending, but the second one will receive a [Check Door]. What could be the reason?

Thanks!
Dave

chamnit, 2016-02-19:
@mariocaptain : The '@' is an undocumented real-time command character. I'll update the documentation shortly.

closed 2016-02-19 by chamnit

GuiRitter, 2019-04-11:
> @mariocaptain : The '@' is an undocumented real-time command character. I'll update the documentation shortly.

Found the updated documentation [elsewhere](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands). May be what [happened to me](https://github.com/grbl/grbl/issues/1267#issuecomment-482291066). Maybe a communication error turned a character into an `@`.