grbl/grbl #912: Sub Programs?

rm-design, 2016-02-21:
Hello,

Does anybody know a trick to get a sub program (M98, M99) to work?
As per grbl M98 and M99 are not available....yet?

109JB, 2016-02-21:
Grbl can't do sub programs.  A gui would have to implement this since grbl basically works from a simple stream of data.

chamnit, 2016-02-22:
@rm-design : @109JB is correct. Grbl does not do macros, variables, or subprograms. There are a number of technical issues with doing this on a system that is streamed to. Particularly loops. However, modern programming languages, like Python, can do the same thing and much more, with a little bit of programming know-how.

closed 2016-02-22 by chamnit

servant74, 2016-02-22:
> <> ... Jack

To have a full implementation, gotta go to a more heavy duty
implementation.  Some like Mach or LinuxCNC (aka EMC2).  Grbl runs on
lighter duty hardware.  That is pretty much the tradeoff IMHO.

For someone into Python, perl or any other regular procedural language,
writing a pre-processor is not that hard.  For the more hard core unix
geeks, even using yacc would be a reasonable implementation too.

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin