grbl/grbl #913: NO and NC combined limit switch support

jhapke, 2016-02-23:
I have hall sensors on X and Y which are active low. Z has optical limits which are active high.  Seems like the $5 setting should be a bit mask so a single axis can be inverted. 
I don't see an option in config.h to deal with inverting only one axis limit switch.
This should be a config option. 
Thanks

chamnit, 2016-02-23:
@jhapke : Thanks. I did something along the lines of an invert mask for the control pins, so you can mix NC and NO switches. I'll do the same thing with the limit pins shortly.

closed 2016-02-23 by chamnit

chamnit, 2016-03-04:
@jhapke : Finally got around to this. You will now see a new config.h option that will allow you mask the limit pins, such that you can have mixed NO and NC switches. Just know that the $5 invert limit pins still applies to all of the limit pins. This mask just toggles the ones you specify.

thejojk, 2017-07-16:
How do I mask different switches? On X+- Y+- and Z+ I have NC switches. On Z- I want a touch plate (NO).

jahnj0584, 2017-07-16:
Check the grbl wiki

thejojk, 2017-07-16:
Sorry, I have looked but can't find anything...

jahnj0584, 2017-07-16:
Many guides on working NC switches. Probe should be NO by default anyways.

electrokean, 2017-07-16:
@thejojk You need to adjust your $5 and $6 config options - https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration
Also look at the different types of G38 probe commands - http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38