grbl/grbl #914: Add empirical PWM->RPM lookup table & smooth spindle control

doppelhub, 2016-02-24:
I propose adding two features to spindle_control.c:
1: Slowly ramp to pwm setpoint.  Most PWM motor controllers will faithfully attempt to instantaneously start or stop a motor, causing massive current inrush.  See lines 139 to 149 for my proposed proof-of-concept.  Since "S" commands are blocking, I don't see any concerns using delay_ms().  Note that this smoothing function ONLY works when the PWM output is active and then only with "S" commands.  For example:

`//motor speed was previously "S10000"
M5
S0`

...stops PWM train immediately because the PWM output is disabled before the speed ramps down. 

`//motor speed was previously "S10000"
S0
M5`

...smoothly ramps PWM duty, as desired.

2: I want the "S" speed to correlate with actual (empirical) RPM.  I've added a lookup table that allows you to alter the output counter register value.  See lines 63 to 98 for the lookup table, which is indexed on line 137.  Note: The LUT values reflect my empirical data... you'd want to comment out line 137 until the user replaced the LUT values.  This lookup table is stored in flash and doesn't consume any ram.

Yay, now my spindle ramps smoothly and doesn't pull 100A when I stop it!

Complete spindle_control.c:

`/*
  spindle_control.c - spindle control methods
  Part of Grbl v0.9

  Copyright (c) 2012-2014 Sungeun K. Jeon

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/
/* 
  This file is based on work from Grbl v0.8, distributed under the 
  terms of the MIT-license. See COPYING for more details.  
    Copyright (c) 2009-2011 Simen Svale Skogsrud
    Copyright (c) 2012 Sungeun K. Jeon
*/ 

#include "system.h"
#include "spindle_control.h"
#include "protocol.h"
#include "gcode.h"

void spindle_init()
{    
  // On the Uno, spindle enable and PWM are shared. Other CPUs have seperate enable pin.
  #ifdef VARIABLE_SPINDLE
    SPINDLE_PWM_DDR |= (1<<SPINDLE_PWM_BIT); // Configure as PWM output pin.
    #ifndef CPU_MAP_ATMEGA328P 
      SPINDLE_ENABLE_DDR |= (1<<SPINDLE_ENABLE_BIT); // Configure as output pin.
    #endif     
  #else
    SPINDLE_ENABLE_DDR |= (1<<SPINDLE_ENABLE_BIT); // Configure as output pin.
  #endif
  SPINDLE_DIRECTION_DDR |= (1<<SPINDLE_DIRECTION_BIT); // Configure as output pin.
  spindle_stop();
}

void spindle_stop()
{
  // On the Uno, spindle enable and PWM are shared. Other CPUs have seperate enable pin.
  #ifdef VARIABLE_SPINDLE
    TCCRA_REGISTER &= ~(1<<COMB_BIT); // Disable PWM. Output voltage is zero.
    #ifndef CPU_MAP_ATMEGA328P 
      SPINDLE_ENABLE_PORT &= ~(1<<SPINDLE_ENABLE_BIT); // Set pin to low.
    #endif
  #else
    SPINDLE_ENABLE_PORT &= ~(1<<SPINDLE_ENABLE_BIT); // Set pin to low.
  #endif  
}

//LUT to find corresponding PWM->RPM register value
//Substitutes requested 8bit PWM counter value with empirical RPM data
	static const __flash uint8_t lookup[256] = {
		0,	0,	59,	67,	73,	78,	82,	86,
		89,	93,	96,	98,	101,	103,	106,	108,
		110,	112,	114,	116,	118,	119,	121,	123,
		124,	126,	127,	129,	130,	132,	133,	134,
		136,	137,	138,	139,	141,	142,	143,	144,
		145,	146,	147,	148,	149,	150,	151,	152,
		153,	154,	155,	156,	157,	158,	159,	160,
		161,	162,	162,	163,	164,	165,	166,	166,
		167,	168,	169,	170,	170,	171,	172,	173,
		173,	174,	175,	176,	176,	177,	178,	178,
		179,	180,	180,	181,	182,	182,	183,	184,
		184,	185,	186,	186,	187,	187,	188,	189,
		189,	190,	190,	191,	192,	192,	193,	193,
		194,	194,	195,	196,	196,	197,	197,	198,
		198,	199,	199,	200,	200,	201,	201,	202,
		203,	203,	204,	204,	205,	205,	206,	206,
		207,	207,	208,	208,	208,	209,	209,	210,
		210,	211,	211,	212,	212,	213,	213,	214,
		214,	215,	215,	215,	216,	216,	217,	217,
		218,	218,	219,	219,	219,	220,	220,	221,
		221,	222,	222,	222,	223,	223,	224,	224,
		224,	225,	225,	226,	226,	226,	227,	227,
		228,	228,	228,	229,	229,	230,	230,	230,
		231,	231,	231,	232,	232,	233,	233,	233,
		234,	234,	234,	235,	235,	236,	236,	236,
		237,	237,	237,	238,	238,	238,	239,	239,
		240,	240,	240,	241,	241,	241,	242,	242,
		242,	243,	243,	243,	244,	244,	244,	245,
		245,	245,	246,	246,	246,	247,	247,	247,
		248,	248,	248,	249,	249,	249,	250,	250,
		250,	251,	251,	251,	252,	252,	252,	252,
		253,	253,	253,	254,	254,	254,	255,	255
		};

void spindle_run(uint8_t direction, float rpm) 
{
  if (sys.state == STATE_CHECK_MODE) { return; }
  
  // Empty planner buffer to ensure spindle is set when programmed.
  protocol_auto_cycle_start();  //temp fix for M3 lockup
  protocol_buffer_synchronize(); 

  // Halt or set spindle direction and rpm. 
  if (direction == SPINDLE_DISABLE) {

    spindle_stop();

  } else {

    if (direction == SPINDLE_ENABLE_CW) {
      SPINDLE_DIRECTION_PORT &= ~(1<<SPINDLE_DIRECTION_BIT);
    } else {
      SPINDLE_DIRECTION_PORT |= (1<<SPINDLE_DIRECTION_BIT);
    }

    #ifdef VARIABLE_SPINDLE
      // TODO: Install the optional capability for frequency-based output for servos.
      #define SPINDLE_RPM_RANGE (SPINDLE_MAX_RPM-SPINDLE_MIN_RPM)
      TCCRA_REGISTER = (1<<COMB_BIT) | (1<<WAVE1_REGISTER) | (1<<WAVE0_REGISTER);
      TCCRB_REGISTER = (TCCRB_REGISTER & 0b11111000) | 0x02; // set to 1/8 Prescaler
      if ( rpm < SPINDLE_MIN_RPM ) { rpm = 0; } 
      else { 
        rpm -= SPINDLE_MIN_RPM; 
        if ( rpm > SPINDLE_RPM_RANGE ) { rpm = SPINDLE_RPM_RANGE; } // Prevent uint8 overflow
      }
      uint8_t setpoint_pwm = floor( rpm*(255.0/SPINDLE_RPM_RANGE) + 0.5);
      #ifdef MINIMUM_SPINDLE_PWM
        if (setpoint_pwm < MINIMUM_SPINDLE_PWM) { setpoint_pwm = MINIMUM_SPINDLE_PWM; }
      #endif
	
	setpoint_pwm=lookup[setpoint_pwm]; //LUT to find corresponding PWM->RPM register value
	
	uint8_t current_pwm = OCR_REGISTER; //get previous PWM register value
	while(current_pwm != setpoint_pwm) {
		if(current_pwm > setpoint_pwm) {
			current_pwm -= 1;
		} else {
			current_pwm += 1;
		}
	
		OCR_REGISTER = current_pwm; // Set PWM pin output
		delay_ms(2);
    }
	
      #ifndef CPU_MAP_ATMEGA328P // On the Uno, spindle enable and PWM are shared.
        SPINDLE_ENABLE_PORT |= (1<<SPINDLE_ENABLE_BIT);
      #endif
    #else   
      SPINDLE_ENABLE_PORT |= (1<<SPINDLE_ENABLE_BIT);
    #endif

  }
}

`

rwensley, 2016-02-24:
I agree completely.  I have been working toward this same idea but time slows me down.  The universal motors are highly non-linear and do not like to be stopped at full voltage.  Both of these options will be a great benefit.

Every router will have a different curve that will need to be put into the lookup table.  Perhaps a small utility that could aid the user in determining that curve and then incorporate it into the spindle_control.c routine.

chamnit, 2016-02-24:
@doppelhub : Thanks for the code proposal. I'll dive right into my thoughts on them. First, delay_ms() absolutely cannot be used with spindle control, because real-time spindle overrides will be broken (written, working, but not published). It's not a good idea to use a blocking delay function, if it at all possible.

Second, you're not the first person to suggest more precise spindle control. Basically my answer was that its not Grbl's job to determine how a spindle should behave. Only that it should output consistent and predicable signals to indicate on/off and the desired RPM. A separate spindle controller should be the one performing the tasks to ensure the spindle is running at the correct RPM and does so smoothly to avoid the amps problem you mentioned.

That said, I recognize that sometimes users have spindle controllers that they can't alter, so the only solution is to alter the spindle control output, as you do in your code proposal. It's a solution, but it's not a general solution that will work for everyone. There are numerous ways of approaching this problem and hence numerous solutions. For now, I think it's still prudent to use a separate controller to interpret Grbl's signals, which can be done by a simple $5-$10 Arduino Mini with a quick Arduino IDE program.

@rwensley : To expand on the idea that every router and universal motors are different and highly nonlinear, this further solidifies the idea that a spindle controller should be separate. A dedicated MCU that can be easily modified, updated, and configured for each particular spindle without risking creating bugs in the main Grbl firmware or overcomplicating things.

doppelhub, 2016-02-24:
Thanks for the informative replies!
Sonny, as long as I don't use real-time spindle overrides, delay_ms() is fine, correct?
I ask because I wrote this code to update the 1500 CNC machines we've already shipped.

chamnit, 2016-02-24:
@doppelhub : For the delay_ms(), I can't guarantee that it won't be a problem, especially in the future with v1.0.  As far as helping you to support 1500 CNC machines, most of my time is invested in supporting official Grbl sponsors and the rest of it is in developing v1.0 and helping users. I can't ensure backward compatibility for you later if you went ahead with this. Nor can I ensure that there will be any room for this code in v1.0, as it's looking to completely use all the remaining flash space in the 328p.

doppelhub, 2016-02-24:
It's short-sided on our part, but we're still using 0.9g and don't plan on updating anytime soon.  I accept the risk.  FYI: We didn't see any issues with this code in our V&V.

Also, what do we need to do to become an official sponsor?  Probably more than the $30 I sent you last year, right ;)?  Probably time for another donation.  As always, thanks for your commitment to grbl!

langwadt, 2016-02-24:
is it normal procedure that you have to manually add dwells to the gcode to take spindle spinup time into account?

doppelhub, 2016-02-26:
Yes.  If your spindle takes a while to spin up, then you'll need to dwell.  grbl doesn't wait at all... just updates the PWM and then executes the next line of code.

csdexter, 2016-02-26:
... which is why in the pro world you have a line called "SPINDLE_AT_SPEED" coming from the spindle controller back into the generic CNC controller to hint the latter it can resume execution after an S word (IIRC the G-Code explicitly mentions that the controller should do that if it can technically detect the current spindle speed).