grbl/grbl #915: arduino due port

bronek999, 2016-02-25:
Tomorow I testet grbl v0.9 cide ported to arduino due. I have 7" lcd with touch and SD card file system.
Now is ported only x,y,z axes. No spindle, limit control, no uart. First speed test config
#define DEFAULT_X_STEPS_PER_MM 3200.0
#define ACCELERATION_TICKS_PER_SECOND 300 
#define DEFAULT_X_ACCELERATION (300.0*60*60)

test code          G1X500F2000
It gives correct 106kHz pulses. with graphical interface 800x480.

Tested with real machine 
320 steps per mm
screw 5mm per  revolution
3 phase stepper motor+driver
with TinyG2 was max possible speed about 3100mm/min
with GRBL it works at 5000mm/min

chamnit, 2016-02-25:
@bronek999 : That's great! I'd love to see the code. It'd help me out. FWIW, Grbl on the 328p has been tested to go as fast as 35,000 mm/min.

bronek999, 2016-02-25:
35,000mm/min? Possible, but I don't think with lead screw.
If I change my screw to 3M belt (20 tooth pulley) it is 60.000mm/min. teoretical speed.

chamnit, 2016-02-25:
@bronek999 : Do you have the code posted anywhere?

bronek999, 2016-02-25:
not right now
I downloaded original grbl yesterday. It is NOT usable version.
[https://www.youtube.com/watch?v=cm6nt1PZ7J0](url)

Moffy, 2016-03-03:
I too would like to see the code, for educational purposes:) I have been investigating the DUE as a platform and it looks good, 100k of RAM, 512k of FLASH and an 84MHz clock with CortexM3 core. After downloading Atmel Studio 7 and looking at their ASF, I found that converting the code I previously wrote for the TM4C123G would be almost trivial because they are basically the same core except the TI chip is M4 with a floating point unit. With CMSIS there is almost a 1 to 1 correspondence in the functions, very cool.

bronek999, 2016-03-04:
Project is available on [https://github.com/bronek999/grbl-arduino-due-port](url).
I made ArduinoDue shield with SD card, RS485 transceiver, DS1338Z for RTC (backup by CR2032 battery), optocouplers for stepper drivers. Now I´m waiting for SMT pinheaders and optocouplers.......
[image]

Moffy, 2016-03-04:
Thanks very much! appreciated.

Moffy, 2016-03-06:
Wrote the serial.c over the weekend for the DUE. Nothing fancy like PMC, just very similar to the ATMEGA code. Unable to test it as I haven't got my DUE yet. Kept it simple like your code no ASF.

bronek999, 2016-03-07:
I have some file with USART initialization and read/write acces for DUE
[https://github.com/bronek999/grbl-arduino-due-port/blob/master/template](https://github.com/bronek999/grbl-arduino-due-port/blob/master/templatel)

It is RS485 spindle control by modbus protocol.
Now it is simple. No circle read buffer, but it is tested on TinyG2, so it works.

chamnit, 2016-03-07:
@bronek999 @Moffy: This is great. Thanks for pushing this forward. This will work perfectly as a template for what I have planned, which includes a lot of refactoring and an overhaul of the entire system. If you are willing, please concentrate your efforts on the low-level stuff. I simply haven't had the time to really dig into it other than using the ASF libraries and haven't looked into whether or not they will perform well.

Moffy, 2016-03-07:
@chamnit Happy to help with the low level stuff. Thanks for the link @bronek999, will have a look, but I wrote for the UART which I guess is a little simpler because you have fewer choices?
@chamnit It might be simpler to avoid ASF as @bronek999 has done with his code. It is easier to follow, as it closely follows the data sheet and the ASF is written for many platforms: AVR8, AVR32, ARM and its structure is VERY complex and ambiguous. You have GPIO, PIO, IO which can get very confusing quickly. That was the nice thing about the TM4C123G, the CMSIS was for only one device and was well documented.

chamnit, 2016-03-07:
@Moffy : Yes, I've run into those very same ASF issues in my time with it. To a non-professional coder, it's been quite a time-sink to try to learn it.  Anyhow many thanks. It'll save me a ton of time and I'll be able to concentrate on the algorithmic and system-level stuff.

bronek999, 2016-03-10:
Added support for SD card read/write. 
It uses free FatFS library. Connection in SPI mode.
[https://github.com/bronek999/grbl](https://github.com/bronek999/grbl)
Moffy thanks for serial.c file

Diy smt pinheaders. Still waiting for optocouplers
[image]

Moffy, 2016-03-10:
@bronek999 Cool. Please find slightly modified serial.c, a few minor changes.
[serial2.zip](https://github.com/grbl/grbl/files/168162/serial2.zip)
Assigning interrupts intelligently could be a significant and beneficial issue.
Board is looking really nice!

bronek999, 2016-03-22:
First file run
[https://www.youtube.com/watch?v=jTld7UcvbvU](https://www.youtube.com/watch?v=jTld7UcvbvU)

SD card works. Grbl works. Having problem with spindle control.  `protocol_buffer_synchronize() `

109JB, 2016-03-23:
@bronek999 

Please ell me about the touch screen system in your video.

bronek999, 2016-03-23:
7 inch resistive touch screen. Connected to TFT display driver RA8875.
Display driver connected to arduino due by SPI interface.
[display](http://www.buydisplay.com/default/7-inch-lcd-module-capacitive-touch-screen-panel-i2c-spi-serial)

109JB, 2016-03-23:
Is the interface code that is running the touch screen on the same DUE that is running GRBL or a separate one?  Thank you.

bronek999, 2016-03-23:
Same arduino due of course

109JB, 2016-03-23:
Very Impressive.

JBSchueler, 2016-05-07:
Nicely done!
Currently I have 2 grbl arduino nano boards running.
One with original clock of 16Mhz and one overclocked to 24MHz just to see the difference.
Unfortunately the 24MHz isn't that faster in computation as expected.

Therefor I would like to move to an ARM as well.
I have an Arduino Due lying around doing noting so I could use that board to do the job.
Another why to get there is to use an STM32F103 development board ( extremely cheap nowadays ), which is a Cortex-M3 like the SAM3X. Downside is that there are no good gbrl ports to that processor.

A Liquid Crystal Display or SDCard interface is a nice to have but I really like to have the old fashion Serial Port to control the machine directly. Does the code still support this feature?
Does it provide PWM for Spindle as well?

The only outputs I need for the cnc-laser are

- X and Y-step
- X and Y-dir
- Spindle enable (PWM controlled)

X3msnake, 2016-05-07:
What about using a esp8266. It is less than 5usd at 80mhz?

No dia sábado, 7 de maio de 2016, Ben Schueler <notifications@github.com