grbl/grbl #916: X-axis movement is reversed after homing

peejster, 2016-02-28:
Homing works as expected. I am only homing the X and Y axis. I home the Y to the highest point and I home X to the lowest point. After homing, if I issue a movement command to the Y axis in the negative direction, it works as expected. However, if I issue a movement command to the X in the positive direction, I get a soft limit alarm but if I issue a movement command to the X axis in the negative direction, it moves.

To home just X and Y, in the config.h, I commented out the default homing cycle for Z and made the homing cycle for X and Y the zero homing cycle:
//#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
// #define HOMING_CYCLE_2

To home Y to the highest point but X to the lowest, I set the homing dir invert mask to 1:
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=1 (homing dir invert mask:00000001)

I don't have any other things inverted:
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)

I want the machine to home in the top left corner (x=0 and y=400mm). I know my directions are correct - I turned off homing, moved the head by hand to a mid X and mid Y position, and then:
1. Entered G0 X2 and the result was movement along the X axis from left to right - as expected.
2. Entered G0 X-2 and the result was movement along the X axis from right to left - as expected.
3. Entered G0 Y2 and the result was movement along the Y axis from bottom to top - as expected.
4. Entered G0 Y-2 and the result was movement along the Y axis from top to bottom - as expected.

Now, with homing activiated, I entered "$H". The X axis moved from right to left, triggered the limit switch as expected (moved left to trigger switch, moved off switch, then slowly moved left to trigger switch again and just come off it), and came to a stop. The Y axis did the same (moved from bottom to top) and came to a stop. Then, I manually entered "G0 Y-2" and the Y axis moved down. Then, I manually entered "G0 X2" but the X axis didn't move and the soft limit alarm was triggered. If, instead, I issued "G0 X-2", the X axis would have moved left to right.

I can't figure out what I am doing wrong with respect to the motion of the X axis after homing.

chamnit, 2016-02-28:
@peejster : Grbl sets the machine coordinates in all negative space. So, the origin is at the rear right corner of the machine. This is standard machine tool coordinate frame setup. Please use the work coordinate systems to offset the working origin to however you like, or you can have Grbl set the origin at the homing point with a  config.h option.

closed 2016-02-28 by chamnit

peejster, 2016-02-28:
Oh. I made the assumption (incorrectly) that the origin would become the location where you homed the machine. (If I left all the defaults for homing (without inverting X) I guess that would be the case.) Thanks for letting me know that by default, origin is rear right corner.