grbl/grbl #917: grbl skips gcode lines during new streams

birkov, 2016-02-28:
I am testing Grbl to run a delta robot with 3 steppers and toshiba driver. It works perfect most of the time however I noticed few strange behaviors, one of which nearly destroyed the robot.
I use python script to stream gcode to arduino (method based on bufferring from steam.py script). 
I prepare a gcode in script of what is supposed to be done (for example 20 lines of code that is run immidiately). Next gcode is prepared and send to arduino after for example 10 seconds. Grbl seems to be skipping some lines (and effectively movement path of end effector changes, which may cause crashing with objects or frame). I noticed this strange skipping of code lines happens only if I use G4 command in the code (or just coincidence?). For example if I have points in space and run code first time after I connect to grbl serial port the effector goes: ABCD. when I run the same trajectory next time it goes ACD, skipping one gcode line. This is done with same serial port connection, same gcode is sent and received (confirmed by grbl response). Am I missing something like gcode commands to end/start current stream? Should serial buffer be flashed everytime gcode trasmission starts? (although it does not seem to be the case). Is there a special method of using G4 commands or reported problems?
Hope the description of the problem is not too complicated to understand.

Second problem I noticed is when I use 5v relay to start/stop 12v vacuum pump it seems to be messing USB/serial connection sometimes. I used separate power supply for the relay, separate power for arduino, made aluminium foil shields for everything and the only clue I have at the moment is USB port in PC is somehow hanging. Did you guys have any experience with similar issue?

chamnit, 2016-02-29:
Grbl was never designed to run a delta machine. Only Cartesian machines. If you made modifications to the code, then I can't help you. If you didn't, it's likely a gcode error being thrown by Grbl by an invalid gcode command or you accidentally altered the serial buffer size when it counts messages sent. The latter would cause data to be corrupted. Or it's a symptom of your EMI problem, where it may be corrupting the data as well.  For this, it usually an improper grounding of equipment or inadequate shielding.

closed 2016-02-29 by chamnit

birkov, 2016-02-29:
I did not modified grbl code, instead I translate delta robot position into cartesian axis steps on pc and stream gcode to grbl. It would not matter if movement is in xyz or discrete delta angles surface. 
Interesting thing is it never skips any line when first cycle is made. It always happen on second or third same stream with G4 codes. If there is no G4 dwells in the program its always good. 
I suspect electrical interefence problems wouldn't always appear in same scenario but be more random.
Here is the code I use to stream gcode to grbl, its pretty much the code from examples (grbl code was not modified).

def write_to_serial_gcode(serial_port, text, debug = ''):
	# Send g-code program via a more agressive streaming protocol that forces characters into
    # Grbl's serial read buffer to ensure Grbl has immediate access to the next g-code command
    # rather than wait for the call-response serial protocol to finish. This is done by careful
    # counting of the number of characters sent by the streamer to Grbl and tracking Grbl's 
    # responses, such that we never overflow Grbl's serial read buffer. 
	#serial_port.flushInput()
	#serial_port.flushOutput()
	RX_BUFFER_SIZE = 128 
	l_count = 0
	g_count = 0
	c_line = []
	for line in text.splitlines():
		l_count += 1 # Iterate line counter
		l_block = line.strip()
		c_line.append(len(l_block)+1) # Track number of characters in grbl serial read buffer
		grbl_out = '' 
		while sum(c_line) >= RX_BUFFER_SIZE-1 | serial_port.inWaiting() :
			out_temp = serial_port.readline().strip() # Wait for grbl response
			if out_temp.find('ok') < 0 and out_temp.find('error') < 0 :
				print "  Debug: ",out_temp # Debug response
			else :
				grbl_out += out_temp;
				g_count += 1 # Iterate g-code counter
				grbl_out += str(g_count); # Add line finished indicator
				del c_line[0] # Delete the block character count corresponding to the last 'ok'
		if debug != '':
			print "SND: " + str(l_count) + " : " + l_block,
		serial_port.write(l_block + '\n') # Send g-code block to grbl
		if debug != '':
			print "BUF:",str(sum(c_line)),"REC:",grbl_out

chamnit, 2016-02-29:
How are you programming the G4 commands? They maybe invalid. Can you also supply a snippet of your problem program to get an idea of what you are sending?

birkov, 2016-02-29:
This is an example doing a square on xy axis waiting for .3sec in every corner.
If I send it first time after I initialize serial port with python its always good. Second and every next is messed up. If I run it without g4 stops its always good. This is just an example but I run into same problem in multiple situation. Is there any particular command to send after each ended block of code?

G1 X500 Y2113 Z998 F160000
G1 X805 Y2275 Z1272 F160000
G1 X500 Y2113 Z998 F160000
G4 P0.3
G1 X1962 Y1553 Z202 F160000
G1 X2144 Y1776 Z515 F160000
G1 X1962 Y1553 Z202 F160000
G4 P0.3
G1 X1962 Y196 Z1559 F160000
G1 X2144 Y509 Z1782 F160000
G1 X1962 Y196 Z1559 F160000
G4 P0.3
G1 X500 Y992 Z2119 F160000
G1 X805 Y1266 Z2281 F160000
G1 X500 Y992 Z2119 F160000
G4 P0.3
G1 X500 Y2113 Z998 F160000
G1 X805 Y2275 Z1272 F160000
G1 X500 Y2113 Z998 F160000

chamnit, 2016-02-29:
Nothing unusual except the multiple feed rates. Unless you are running in inverse time mode, you only need to send a be one when it changes. If you disconnect your Arduino, so it's only Grbl running, does the stream behave the same way? This would rule out your electronics.

chamnit, 2016-02-29:
Also is your step idle delay set at something other than 255?

birkov, 2016-02-29:
I tried with different usb controller and added M2 command at the end of each stream and problem seem to be gone. Also relay is no longer resetting the usb. Thanks for help

chamnit, 2016-02-29:
@birkov : Ok. Thanks for the update.