grbl/grbl #918: Inverting PWM signal and using +5v to switch off spindle motor.

casiobearing, 2016-02-29:
I've just upgraded my 6040 CNC controller with a Arduino Uno running GRBL v0.9j.

All went well but the spindle speed control was working backwards; S0 => full speed, S1000 => stopped, and M5 made the spindle go 100% speed instead being off.

I did some experimental code to invert the PWM signal, and to allow spindle off via M5 with an +5v signal - the easiest way to do this was to switch the PWM port back into normal mode and set the port bit.

I removed the following line from spindle_init()
      SPINDLE_PWM_DDR |= (1 << SPINDLE_PWM_BIT); // Configure as PWM output pin.

spindle_init() probably could do with also setting the pin to the "off" state as per the stop code below to keep the code consistent.

I added the following to the beginning of spindle_stop()

  #ifdef VARIABLE_SPINDLE
    SPINDLE_PWM_DDR &= ~(1 << SPINDLE_PWM_BIT); // Configure as normal output pin.

#ifdef INVERT_SPINDLE_ENABLE_PIN
    SPINDLE_PWM_PORT &= ~(1 << SPINDLE_ENABLE_BIT);
#else
    SPINDLE_PWM_PORT |= (1 << SPINDLE_ENABLE_BIT);
#endif

I added the following line into spindle_set_state() - (to switch the port into PWM mode)

SPINDLE_PWM_DDR |= (1 << SPINDLE_PWM_BIT); // Configure as PWM output pin.

TCCRA_REGISTER = (1<<COMB_BIT) | (1<<COMB_BIT0) | (1<<WAVE1_REGISTER) | (1<<WAVE0_REGISTER);

plus a #define COMB_BIT0 COM2A0

Does that sound like a sensible option to add to the config options?

viewsat, 2016-03-20:
Hello casiobearing:
Can you give me the modified spindle_control.c this file?
thank you!

lalo-uy, 2016-03-20:
the simples way is to change the pwm value from x to 256-x that will invert
the pulse .

casiobearing, 2016-03-20:
Didn't want to mess with the complicated pwm calculation stuff buried in there so the simplest was to invert the output was to flip the PWM register bit, and to make the port a simple I/O for stopping.
Modified code supplied to viewstat.

viewsat, 2016-03-21:
Hello casiobearing,
Yesterday you email to me modified code, let pwm voltage can be inverted.
Thank you very much!

marcogarzotti, 2016-07-14:
Hi all,
i arrived here searching a way to control properly my laser driver, that works with a negate logic respect to a standard spindle.
I think that this could be a very useful upgrade for a lot of people that are using a laser driver!

@casiobearing: i'll try to modify mi files as you wrote.
Can you send also to me your modified file? Thank you.

viewsat, 2016-10-26:
Hello casiobearing:
Can you modified GRBL V1.1  inverting PWM signal and using +5v to switch off spindle motor.
thank you!