grbl/grbl #920: Alarm condition on power up

pihnat, 2016-03-03:
Hi,
I'm using 0.9j and have an alarm condition when I power up the Arduino. The docs say that this should only happen if Homing is enabled but I have Homing turned off ie parameters $20 -> $23 are all 0, $24->27 are default values. Note that this doesn't happen with 0.9i.

109JB, 2016-03-03:
There is a setting in config.h that could be set to do that. Check it to see. Here is what it says in config.h

// When Grbl powers-cycles or is hard reset with the Arduino reset button, Grbl boots up with no ALARM
// by default. This is to make it as simple as possible for new users to start using Grbl. When homing
// is enabled and a user has installed limit switches, Grbl will boot up in an ALARM state to indicate 
// Grbl doesn't know its position and to force the user to home before proceeding. This option forces
// Grbl to always initialize into an ALARM state regardless of homing or not. This option is more for
// OEMs and LinuxCNC users that would like this power-cycle behavior.
// #define FORCE_INITIALIZATION_ALARM // Default disabled. Uncomment to enable.

pihnat, 2016-03-03:
If I read it correctly it implies with a fresh install of 0.9j and with Homing disabled the default behaviour is to boot up with **no ALARM** to make it simple for new users. Only after you change config.h and re-compile it as OEMs and LinuxCNC users would do should it boot with ALARM on.

However with a fresh install of 0.9j and Homing off it's doing the opposite ie booting with Alarm on not off.

milezdr, 2016-03-03:
You need to restore all values to  default in grbl with command and after that make fresh install

pihnat, 2016-03-03:
Already tried that, It didn't work. I still get ALARM at power up.

chamnit, 2016-03-03:
@pihnat : Can you send a print out of your Grbl settings? Also, if you disconnect everything from the Arduino except the USB cable, does it still ALARM upon power up?

pihnat, 2016-03-03:
I'm running it on my desk with just the USB connected. It's a Funduino (Uno). Here's the settings:
[image]

chamnit, 2016-03-03:
@pihnat : Well I'm baffled. How are you installing Grbl? Via the Arduino IDE method with the source code on this site? And please provide the build date. You can get it by sending Grbl a $I command. You should get a version string.

pihnat, 2016-03-03:
[0.9j.20151218:]
I just download the hex file from the main page of this site under Master Branch:
      Grbl v0.9j Atmega328p 16mhz 115200baud with generic defaults (2015-12-18)
I install using XLoader. Note that 0.9i doesn't have this problem.

chamnit, 2016-03-03:
@pihnat : Ok. Verified the bug on both the source and build firmwares. Not sure why it's doing this, but it had to do with a 'sizeof()' when it initializes the variable. I'll push the fix later tonight.

closed 2016-03-03 by chamnit

chamnit, 2016-03-04:
@pihnat : The alarm upon startup fix has been uploaded. Please try it out.

pihnat, 2016-03-04:
Fantastic, works as expected now. Thanks for all your efforts.
regards
Peter

rajpalvishwakarma, 2018-09-07:
Hi All.
One more solution for the Alarm error.   JUst need to go Advanced setting > click on  Soft reset Grbl.