grbl/grbl #921: supporting 3d printing

qingzhengzhuma, 2016-03-03:
I'm working with the 3d printing，I‘ve read the README file and know that the GRBL is not support the E Non-command word， but the Gcode file I generate from STL file (using Slir3c and KISSSlicer) contains the E word, can anybody give me some tips about slicing tools or GCode interpreter for supporting 3d printing?

milezdr, 2016-03-03:
You can't use grbl for 3d printing because for now we don't have 4 axis for the extruder. Only we can mimic x, y, z we need more pins for extruder

qingzhengzhuma, 2016-03-03:
thank you so much！@milezdr

vlachoudis, 2016-03-03:
I was wondering, the extruder moves only in one direction, could the spindle pins be used to send steps to the extruder?

Norwegian-Gadgetman, 2016-03-03:
No.

The extruder needs to be able to move at different speeds, and in both directions.
And it needs to do so accurately. 
A normal 1.7mm filament has a cross-sectional area of 2.41mm2.
When you lay down a 0.1mm high, 0.4mm wide track, you get 60.25mm of track for every mm of filament that goes into the hot-end.
And every time there's an interruption of that track, the extruder needs to 'retract' filament to avoid stringing.

vlachoudis, 2016-03-03:
Ok I see, I do the same with the hot-glue gun I retract the plastic every time in the end :)

I don't have a 3D printer for the moment, I am asking since I am interested in building one and drive it with grbl as well.

Still there are the pins 13 Spindle direction and 11 Spindle PWM, that could be used to give steps and direction to the extruder. Plus the A3 Coolant enable, A4 not used/reserved. Does the 3D printer needs more pins? (hotbed, temp sensor etc?)

lalo-uy, 2016-03-03:
Or set it to PWM , it will be the extruder feed rate.

alfredanil, 2016-03-03:
Uno does not satisfy "advanced" 3D printing demands due to less number of pins. Would be more logical if Grbl is focussed on non 3D printing applications. Teacup is a firmware which can fill this position for Uno like boards for basic print jobs.

qingzhengzhuma, 2016-03-03:
So glad to see your comments， and i find a solution. it seems to work, i'll try. http://www.instructables.com/id/Arduino-Controlled-CNC-3D-Printer/?ALLSTEPS

closed 2016-03-03 by qingzhengzhuma

chamnit, 2016-03-03:
FYI, Grbl is strictly a 3-axis motion controller. Its goal is to solidify the motion controller foundations for which other machine types can be built upon. Unfortunately that means restricting additional functionality, such as extruders for the time being. However, Marlin is a direct Grbl descendant and is design for 3d printers. Please look into that project, if you don't know what it is.

That said, I do have plans for eventual 3d printer support in the next generation of Grbl, but that is still a ways away.

qingzhengzhuma, 2016-03-04:
Greate! it helps a lot. I've read the wiki of Marlin, that's a good candidate~@chamnit

ryannining, 2018-02-19:
Uno and nano can do 3d print, look for github teacup and github karyacontroller they both support nano, sdcard 3d printing

X3msnake, 2018-02-19:
You can tryout Repetier Firmware it allows to build multitool machines with
the same mega

ryannining, 2018-02-19:
you cant install repetier, marlin on nano or uno, the flash and ram is not enough.

X3msnake, 2018-02-19:
@ryannining Yes only on the mega as I stated on the comment :)

simrim1, 2022-01-02:
Hi,
I don't know if this thread is relevant. @chamnit has indicated that there is a future plan for 3d printer support so I was thinking baout get rid of limit switches for 3d printing. Usually 3d printers use belts to move around axis which are very easy to move by hand. So, it should be the user's responsibility to home the printer manually and with the use of soft limits we can use grbl effectively for 3d printing. To be on the safe side this grbl version will have predefined soft limits of 15-20 mm which must be redefined by the use depending on their printer's capability of lengths of movement. I believe it will still be a success because Arduino based on ATmega328x are the cheapest options.

@chamnit can this be a practical solution?

Thanks