grbl/grbl #922: Connect Grbl to ZigBee for wireless transmission

makeacoffeeprj, 2016-03-03:
Hi everyone! You're doing an amazing job and I'm wondering if G-Code can be streamed "wirelessly", maybe with a ZigBee node attached to the USB port of an Arduino with an adapter (i.e. the XBee Explorer USB from Sparkfun - https://www.sparkfun.com/products/11812).

I read all the section about writing an interface, so I am sure thant I can send a g-code string through zigbee protocol (https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#writing-an-interface-for-grbl).

Does anyone know if it's possible for GRBL to read from a ZigBee node connected via usb adapter to the arduino?

Thank you so much,

Diego

Norwegian-Gadgetman, 2016-03-03:
The Explorer USB is designed to be hooked up to an USB Master, such as a PC, not to a 'slave' device'.
The SparkFun Xbee Shield is probably more what you need on the Arduino.

chamnit, 2016-03-03:
@makeacoffeeprj : Wireless communication is something I'm looking into for the next generation of Grbl. The 328p won't be able to handle much more than what it is doing now. So your current option is to have another micro controller/microcomputer stream Grbl g-code and act as a wireless hub for the computer. Not ideal, but that's how OctoPrint essentially works.

makeacoffeeprj, 2016-03-03:
Hi @Norwegian-Gadgetman: I think you are right but the XBee Simple Explorer would have allow me to  not modify the Grbl code that is expecting to read from the USB. The XBee Simple Explorer acts as a serial-to-usb converter. Am I right if I say that using the XBee Shields is not possible with the current Grbl code?

@chamnit: I already developer all the stack to send whatever kind of string from a Raspberry Pi via ZigBee using REST Services. The basic point is this string can be read from a ZigBee node. What do you mean with "The 328p won't be able to handle much more than what it is doing now"? Sorry but I'm a newbie.

lalo-uy, 2016-03-03:
You can check the ESP8266 modules.
It easy to get a Wifi serial port extender with them, an are way cheap in
asia providers.
Look for ESP-01 or better ESP-12E.

chamnit, 2016-03-03:
@makeacoffeeprj : I simply meant that if you are altering the Grbl source code, you'll likely run into problems. If you are just streaming data to Grbl, as you normally would, I don't see any issue with connecting GRbl to a wireless adapter. If you have an option to do so, I would wire your host directly into the Arduino's RX-TX serial pins. You'll avoid USB altogether in this way.

makeacoffeeprj, 2016-03-03:
@chamnit so Grbl is already set up to read from the rx tx serial pins, great news! One last hint please: do you know any cnc that are already-made sold and that can be driven by arduino uno board hosting Gbrl? I want to focus more on the sw than one figure out how to build one of them. I'm not really interested in a plotter o a laser cutter, I want to use a tiny water pump.

109JB, 2016-03-03:
I have an ESP8266 and have been meaning to test this mode. I may give it a try tonight.

makeacoffeeprj, 2016-03-08:
Any news?
Do anyone know any cnc that are already-made sold and that can be driven by arduino uno board hosting Gbrl? I found this product from makeblock but they have their own firmware (http://www.makeblock.cc/xy-plotter-robot-kit/)

Norwegian-Gadgetman, 2016-03-08:
You could pick up a ShapeOko clone, probably. 
The eShapeOko (from England) komes without electronics. 
Inventables' X-carve is also based on the same design, comes with the electronics, including an Arduino.
The ShapeOko 3 comes with its own electronics, too.
There may be a ShapeOko 2 kit laying about some place if you check the buying and selling area of www.shapeoko.com/forum  
A First Gen ShapeOko may be cheaper, but it requires a bit extra to get it to perform properly. And they're probably well on their way to become collectors items by now.

The MakerBlock kit looks kind of spindly to me.

SloMusti, 2016-03-08:
@makeacoffeeprj we have https://github.com/IRNAS/PreciseXY that can be sorted out as a kit for you and runs grbl from arduino uno. Might be a convenient and cheap option for you to do software testing if the fact that it uses CoreXY approach does not interfere.

makeacoffeeprj, 2016-03-09:
@Norwegian-Gadgetman @SloMusti  I'm looking for an xy (and maybe z) axis plotter - the cheapest one - and substitute the carver/pen/laser with a water pump. I gave a look to your suggestions but still they're too expensive for me, I was looking for something about 300 euros.

schlotzz, 2016-03-09:
@makeacoffeeprj Maybe the CoreBeam laser cutter gives you a platform to use as start for your project. It's a cheap CoreXY platform using 8mm smooth rods, LM8uu linear bearings and GT2 belts. Most parts can be printed in 3D and all other sourced on eBay etc.
Please have a look at http://github.com/schlotzz/CoreBeam

blinkenlight, 2016-03-10:
@makeacoffeeprj The [eShapeOko kit](http://store.amberspyglass.co.uk/eshapeoko-cnc-milling-machine-mechanical-kit.html) might be of interest to you at about 340 euros, but do keep in mind you'll still need a few dozen euros worth of driver electronics...

X3msnake, 2016-03-12:
Tou have a proxxon kit also, that comes with no controller

http://www.proxxon.com/en/micromot/24360.php?list

GabeHC, 2016-04-06:
ESP8266 is a much powerful MCU then AT-MEGA 328, it is 32bit core 4M rom run at 80MHz even 160MHz, and it is dirt cheap. do you have any plan port GRBL on to it?

lalo-uy, 2016-04-06:
ESP8266 will need some kind of port expander, don have to mani i/o pins

ser007, 2016-06-30:
I believe this may be a solution, right?

http://creatorbot.com/projects/electronics/add-esp8266-wifi-to-your-3d-printer

or 

https://hackaday.io/project/7826-esprint

HuubBuis, 2016-10-19:
I connected my arduino GRBL controller to a HC-05 bluetooth module. Windows (10) assigns a ComPort that automaticcally connects to the bluetooth modele when comport is opened by (any) a serial program. The software i am writing uses bluetooth and is running on a 7 inch windows 10 tablet.
Tested using bCNC, TeraTerm, CoolTerm, Bluetooth serial terminal.
p.s. You need the comport with the 'Dev-B' suffix. If you got any problems, disconnect the Bluetooth module, remove it in the bluetooth devices list, wait until the module is not showing up anymore, connect the modele en connect it  to Windows again. Check the comportnumber, because it is probably changed.

GabeHC, 2016-11-14:
ESP32 is out, More GPIO to use.
https://pbs.twimg.com/media/CxBr751WgAEj90T.jpg:large

whitetd, 2016-11-14:
ZIGBEES are really only ment for short packet data, like TV control codes.
A full wifi system would be much better, still think a raspberry Pi acting
as a local buffer/ direct connected to the end controller would be best,
then anything with a browser could drop files, and run, even possible to be
a way to add SD support.

mkeyno, 2016-11-14:
I suggest use the serial bridge with ESP8266 module , check this successful lib by dear Luce https://github.com/luc-github/ESP3D , easily can bind with any interrupter

HoangMaiDat, 2017-08-15:
I successfully connected the zigbee module with arduino nano, but the data transfer failed, because the transfer rate did not respond, it took a lot of data