grbl/grbl #924: Modal Group Errors - Fusion 360

mecizeldec, 2016-03-04:
I'm getting this constantly.

Hardware: RASPI 2, Protoneer CNC Hat (GRBL on Nano)
Software: BCNC (Raspi Dist.), Fusion 360

I've been beating my head against the wall (and google, forums and anything else even remotely similar) for two days over what post processor to use, as anything that seems to make sense keeps throwing these Modal Group Errors. Does anyone know offhand how to get past this with Fusion 360? First Homebuilt CNC and i'm soooo close to my first cut. 

Thanks!

mecizeldec, 2016-03-04:
It's right off the hop. Example of the first one:
G54
M9
G0X-117.4194Y-10.384Z25.6889
error: Modal Group violation
ok
ok
ok
ok
ok
ok
ok

109JB, 2016-03-04:
Post the g-code, not just the responses from grbl.

chamnit, 2016-03-04:
@mecizeldec : Yes we need to see the g-code. A modal group violation occurs when there are two commands in the same modal group in the same line. For example, G0, G1, G2, G3 are all in the same modal group because they are all motion commands. See modal group definitions to see what they are.

mecizeldec, 2016-03-04:
[tEST bOARD.zip](https://github.com/grbl/grbl/files/158049/tEST.bOARD.zip)

chamnit, 2016-03-04:
@mecizeldec : Running it now without any modal group errors. Only a couple of unsupported command errors with the '%' and 'T1 M6', which don't do anything. Now at line 600.

mecizeldec, 2016-03-04:
SO I guess a better question is, what am I doing wrong then? lol.
Is there a config setting that could cause this?
Model orientation problem?
ID10T error?

109JB, 2016-03-04:
Is the file you posted the same as generated the errors. The reason I ask is because in your second post you listed:

G54
M9
G0X-117.4194Y-10.384Z25.6889
error: Modal Group violation

I searched the "tEST bOARD.nc" file you sent and the G0X-117.4194Y-10.384Z25.6889 line is not there.

mecizeldec, 2016-03-04:
Thanks for the response everyone. I was observing some other weird behavior with moving the gantry around. A bit of history:
When I first received the full set of boards, I momentarily shorted out one of the Pololu Drivers (trying to set current limit). This blew up two of the motor drivers and cooked a motor. I suspected that the interface board was also damaged but when I replaced the motor controllers, everything was back up and moving again. I knew things wern't 100% due to some crosstalk noise I could hear on inactive motors during a single axis movement. I was hoping it was just some leakage, and nothing to be worried about. Looking back at some of the issues now, it all seems to point to the board. Even today when I was homing the gantry, the x axis was stuttering during a long move and I was getting homing errors all over the place. When it finally homed in, the X motor controller got really hot, like it was in stall mode. I'm pretty sure part of the communication circuit between the Arduino and the PI is damaged and noisy. When I was getting the Modal errors, It seemed to be in different places in the code at different times. I have ordered a replacement board to see if that clears up my problems. 

Thank you all for your help!

chamnit, 2016-03-04:
@mecizeldec : Thanks for the update. Good luck.

closed 2016-03-04 by chamnit