grbl/grbl #925: Probing routrines

milezdr, 2016-03-04:
Hi Chamnit you and community have done awesome work on grbl . I was wondering what it takes for implementing this probing routines for edge finder and hole finder  to the next version of grbl http://www.timeguy.com/cradek/01262579508
This is "your" job to implement on grbl or the GUI software thing ?

chamnit, 2016-03-04:
@milezdr : In general, it's the GUIs job to perform these tasks. Grbl simply provides the basic tools to do so, which it does with the G38.x probing commands. It's very difficult to support every unique scenario per application. bCNC is a great GUI frontend that allows users to write a plug-in in Python. This would be a great addition to their tools.

closed 2016-03-04 by chamnit

rwensley, 2016-03-04:
I too am having an issue with the probing commands however mine is slightly different.
After making the comms connection I run the program
G21
G38.2 Z-0.5 F5
G92 Z0.65

The Z - Axis Moves AWAY from the work not towards as it should.
I touch the plate to the bit and the Z-Axiz stops and the 0.65mm is entered.
I run the probe program again now the Z-Axis moves toward the work.

Any help in my lack of understanding this will be appreciated.

Ron Wensley

milezdr, 2016-03-05:
@chamnit I asked the same thing in https://github.com/Denvi/grblControl  GUI and Denvi said  "neither grbl firmware nor grblControl app doesn't support RS274/NGC parameters (variables) now."

chamnit, 2016-03-05:
@milezdr : No it doesn't, but parameters/variables are easy to simulate with any modern programming language (arguably much better). bCNC does that with their Python-based GUI. If you can program Python, then you can install a plug-in (macro program).

vlachoudis, 2016-03-05:
@milezdr Just one small correction. You don't need to know Python to write such routines. bCNC accepts parametric code, where each expression can be either written in NGC style or Python style  and assign it to user defined buttons.

Hole finder is already integrated in bcnc, and probing to detect edges is very simple from the interface.

Plugins are more advanced to generate or manipulate the gcode, and for them you need Python knowledge as @chamnit mentioned.

milezdr, 2016-03-06:
@vlachoudis I don't have knowledge at programing.. :/ I saw that bCNC program  but I have very hard time making to work with 64 bit windows because I can find 64 bit pyserial or maybe I don't install it properly.So in order to use bCNC I need to install 32 bit version windows

vlachoudis, 2016-03-06:
@milezdr  I am working mostly on linux, but I have a windows 64bit and the installation went through without a problem with the following command (Look the wiki [https://github.com/vlachoudis/bCNC/wiki/OS]())
```
C:\PythonXX\Scripts\easy_install pyserial
```

milezdr, 2016-03-06:
@vlachoudis I got it working now. The problem was I was using 64 bit python with 32 bit pyserial (because there's no 64 bit pyserial) now I'm using 32 bit python with pyserial 32 bit.
But I find it bCNC little slow. I'm using python 2.7 do you think that maybe the reason for that?What version of python you are using?

vlachoudis, 2016-03-06:
I am using Python 2.7. If you have a huge gcode file the rendering of the gcode can be slow., due to Python being an interpreter language. You can disable it on to of the canvas. However streaming should not be slow at all.