grbl/grbl #926: A Probing Question

rwensley, 2016-03-04:
Greetings,

I am having an issue with the probing commands however mine is slightly different than the recent post.
 After making the comms connection I run the following program
 G21
 G38.2 Z-0.5 F5
 G92 Z0.65

The Z - Axis Moves AWAY from the work not towards as it should.
 I touch the plate to the bit and the Z-Axiz stops and the 0.65mm is entered.
 I run the probe program again now the Z-Axis moves TOWARD the work.

Any help in my lack of understanding this will be appreciated.

Ron Wensley

langwadt, 2016-03-05:
what was the z position before the probe? G38.2 Z-0.5 means move towards Z = -0.5 if the Z was at say -1 before the probe command if would move away from the work

rwensley, 2016-03-06:
Ya know...I really hate it when I miss the obvious. 

Thanks,
Ron Wensley

closed 2016-03-07 by chamnit