grbl/grbl #928: Universal driver GRBL Arduino Zero port

VojislavM, 2016-03-07:
Hi, everybody.

I am working on new version of [Universal Unipolar Stepper Controller](https://github.com/IRNAS/UniversalUnipolarStepperController), and I want to port GRBL to new ARM platform. These are requirements and conclusions for the new version of the Stepper Controler: [Version2](https://github.com/IRNAS/UniversalUnipolarStepperController/blob/master/Universal_Unipolar_Stepper_Controller_V2_review.md). After searching I realized that it is not possible to find a microcontroller with all that features that cost under 3 USD. 

After thinking, I decided to port it to Arduino Zero platform, mainly because I think it is the fastest way to do it, and Arduino Zero microcontroller has all the features I need.

If someone have done this before please share your thoughts, suggestions and repositories.

One thing that worries me is will [Arduino Zero](https://www.arduino.cc/en/Main/ArduinoBoardZero) emulated EEPROM of 16KB be enough for GRBL?
I read that GRBL team is working on GRBL port for ARM. Does somebody know when will that be done?

Thanks in advanced,
Vojislav

chamnit, 2016-03-07:
@VojislavM : Grbl runs very well on a 32KB flash, 1KB EEPROM, and 2KB ram micro controller. So with 256KB flash, 16KB EEPROM, and 32KB ram on the Arduino Zero's ARM, you shouldn't have any problems with capacity. That said, I'm working toward a SAMD21 port of Grbl, but it's a ways away still. Any work toward this goal (as well as the Arduino Due) is most appreciated.

VojislavM, 2016-03-07:
Thank you for the answer. It was my mistake, I thought that AVR has more than 16KB of EEPROM, now I see that it is 1KB. Sorry for stupid question :)

Once I start to port firmware, I will put repository link here so everybody can see it.

Cheers.

langwadt, 2016-03-07:
STMs nucleo is twice as fast and 1/4 the price of arduino zero http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1847/PF260320

I have a port of GRBL running on a nucleo stm32f411 with three of these drivers: http://www.st.com/web/en/catalog/tools/PF260715

chamnit, 2016-03-07:
@langwadt : Oh I don't doubt the STMs are more powerful at all. It's more whether that extra power is needed and if there is a certain level of ubiquity/community acceptance. Even at this point of today's ARMs, it's still not clear which one to pick and why I should pick one over another. In any case, I've been compiling a library of every Grbl ARM port I've run across. This includes your STM port and other Freescale(Teensy) ports.

langwadt, 2016-03-07:
I don't think there is a reason to pick one over the other, they all have similar features. I just happen to use STM and a stack of four boards at ~$10 each is all you need for a 3 axis CNC. 

Using the opencm3 firmware lib it might even be possible to make it somewhat portable between different manufacturers

VojislavM, 2016-03-08:
@langwadt I saw your repository [nucleo_grbl](https://github.com/langwadt/nucleo_grbl), if it is working well as you sad I will go to the STM side because I need to finish this fast. :) I think it would not be a problem to port your code to some smaller STM microcontroller, like [NUCLEO-L053R8](http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1847/PF260001#) that is more suited for this application. 

Can you please share what compiler did you use, and other things to get me started.
I also saw that [NUCLEO-F411RE](http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1847/PF260320) does not have EEPROM, and functions for reading and writing from and to EEPROM are empty [nucleo_grbl/eeprom.c](https://github.com/langwadt/nucleo_grbl/blob/master/eeprom.c) (code is commented). How did you handle that?

langwadt, 2016-03-08:
@VojislavM I've just added a first try at simulating an eeprom with a sector of the flash and it seems to work (15min of code and a few hours of git...) 

get: 

https://github.com/langwadt/nucleo_grbl 
https://github.com/langwadt/libopencm3

install: 
msys - http://sourceforge.net/projects/mingw/files/MSYS/Base/msys-core/msys-1.0.11/MSYS-1.0.11.exe
and
arm-none-eabi/arm-elf toolchain (for example this one https://launchpad.net/gcc-arm-embedded)
you might have to add path to the arm tool chain

run "make bin" in the nucleos_grbl dir to build

when you add the ST driver the nucleus board shows up as a uart and a disk drive
just copy the stm_main.bin to the ST drive  to program

if you can wait a few days I'm porting 1.0c to System workbench for STM32, libopencm3 seems to be much more trouble than it is worth and System workbench is a single install with proper debug based on the eclipse GUI

VojislavM, 2016-03-09:
@langwadt Thanks for instructions, I will try this. 
Also, I will install System workbench, so I can test your new code in few days.
Please send me repository link when you upload your now code for System workbench.

langwadt, 2016-03-15:
@VojislavM new repository at https://github.com/langwadt/grbl_stm32
install system workbench, import project and you should be good to go

VojislavM, 2016-03-23:
@langwadt  Great, thanks. I ordered [NUCLEO-F411RE](http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1847/PF260320) and I will try it as soon as it comes. :)

Andypos, 2016-03-28:
@langwadt  Nucleo411 is very good concept. I port all to Coocox ,and compile .Work ok ,but
for 100MHz I must enable HSE and HSE BYPASS. How You use  SPI ? 
regards

jsmithcarlsbad, 2016-06-12:
Got Grbl 0.9j running on Arduino Zero tonight... this is a step towards porting to new hardware platform that will be available *soon*. New platform uses SAMD21J18 - is there any interest in "official" support for the SAMD21 platform ?

chamnit, 2016-06-12:
@jsmithcarlsbad : Yes. Support for the SAMD21 series is on the table. While porting Grbl directly to ARM will make it faster, it won't make Grbl perform better. Grbl-ARM is a totally new firmware that overhauls just about everything and will improve performance across the board. I will not get into much more detail than that. However, please post your Grbl SAMD21 port. I'd like to see your approach to porting. I always learn a bit from every port I see.

wrljet, 2016-06-16:
Will the STM32 port run on a NUCLEO-F401RE, or only the NUCLEO-F411RE?

langwadt, 2016-06-17:
@wrljet It should run on the NUCLEO-F401RE when build for 84MHz and the slightly smaller RAM

wrljet, 2016-06-21:
@langwadt What is required to use the EEPROM simulation code?
I'm getting:  error: EEPROM read fail. Using defaults

Bill

wrljet, 2016-06-21:
@langwadt Sorry to be bothersome...
Does your port for the X-NUCLEO-IHM01A1 work if these are set?
// #define STANDARD_GRBL
#define VARIABLE_SPINDLE
#define PWM_SPINDLE

I don't seem to get a PWM signal coming out on any pin when I issue M3 to start the motor.
Debugger shows this gets called in splindle_control.c (line 275):

#ifdef VARIABLE_SPINDLE
#ifdef PWM_SPINDLE
	  TIM_SetCompare1(TIM3,current_pwm);

Thanks, Bill

langwadt, 2016-06-22:
@wrljet you'll get EEPROM read fail the very first time you program a board after that it just works
I've updated the code, there was an error so PWM_SPINDLE didn't work unless it was grbl pinout

you should probably put issues with my port on https://github.com/langwadt/grbl_stm32 it isn't really GRBL specific

wrljet, 2016-06-22:
@langwadt OK, will do.  I'm new to this github commenting world.  :-)
I did find and fix the small bug also.  I'll report over on your hub.

Bill

closed 2016-09-03 by chamnit

aqibi2000, 2018-12-03:
Any news on this port?