grbl/grbl #930: Motion problems in circles and lower feed rates

Hogie1982, 2016-03-10:
Hi there,
Recently, I build a small 3axis CNC machine using Grbl, and generally all is working fine. The maximum feed rate is around 400mm/min in X & Y direction and roughly 50mm/min along the Z axis. So happy me :) and congratulations on your work folks.

After doing some test pieces I discovered an issues with motions in a particular feed rate range. When the feed is relative low F35-F42 the stepper motors show erratic movements. This was observed when doing some circles and diagonals at F100. Ie when doing half a circle with radius 10mm in the XY-plane from x0 y10 to x0 y10 and looking **only** at the X-axis, the feed rate of the X-axis goes from its maximum feed (at y10) to a minimum of F0 (at y0), and the back to its maximum (y-10). During this feed rate change it crosses the mentioned feed rate range, in which the stepper makes some erratic movements. 

Generally, one would argue that the problematic range is to be avoided. So just use higher or lower feed rates, but this is impossible for circles, and one has to be careful when choosing feed rates for diagonals.

The question now is, whether anyone has experienced similar issues? And if yes, do you have any idea what this problem could be, and how this could be solved?

I did some test which are describe below and the setup (incl Grbl settings) is summarized thereafter.

**Tests:**
The problematic feed rate  range was defined with a number of tests (please see G-commands below):
First, it was tested whether it is a mechanical problem by using steppers motors which are currently attached to the machine and running a number of motion commands, and then the tests were repeated with the same type of stepper motor, but not connected to the machine (so just laying on the table). The results for the both tests were the same.
Tested G-commands (starting at x0 y0 for linear commands and relative to x0y0 for circular commands):

1. G01 movements along X-axis (ie G01 X10 F50), feed range 5mm/min up to 200mm/min
2. G01 movements along Y-axis (ie G01 Y10 F50), feed range 5mm/min up to 200mm/min
3. G01 45 degree diagonal movements along X & Y-axis (ie G01 X10  Y10 F50), feed range 5mm/min up to 200mm/min
4. G01 movements along Z-axis (ie G01 Z10 F50), feed range 5mm/min up to 50mm/min
5. G02 & G03 movements in XY-plane (ie G02 X10 Y0 I10 J0 F50), feed range 5mm/min up to 100mm/min

**Results:**
The results were as follows:
1)  G01 movements along X-axis (ie G01 X10 F50): 
- Feed rate from F5 to F35: no problems
- Feed rate from F35 to F42: erratic movements
- Feed rate from F45 to F200: no problems 

2) G01 movements along Y-axis (ie G01 Y10 F50): same as for X-axis

3) G01 diagonal 45 degree movements along X & Y-axis (ie G01 X10  Y10 F50):
- Feed rate F5 to F50: no problems
- F50 to F60: erratic movements of both axes (actually looking at X and Y axis separately, these feed rates correlate to F35 up to F42 per axis by multiplying with sin(45 degree))
- above F70: no problems

4) G01 movements along Z-axis (ie G01 Z10 F50): similar to linear movements along X or Y axis

5) G02 & G03 movements in XY-plane (ie G02 X10 Y0 I10 J0 F50):
- Feed rate F5 to F30: no problems
- F35 to 40: erratic movements of when one of the axes is in range of its maximum speed (see description above)
- F45 to F100: some erratic movements of each axis during the circle (see description above)

**My Setup:**
- Arduino Uno R3
- Grbl v0.9j ($I=0.9j.20151218)
- Universal Gcode sender (Version 1.09)
- Arduino CNC Shield v3.00 (Protoneer)
- Stepper drivers: Pololu DRV8825 Stepper Motor Driver Carrier, High Current
- Stepper motors: 400 steps/rev (42byghm809) Nema 17 stepper motor

It was carefully checked that no power cables cross each other, or cross the USB cable from PC to Arduino. Also additional ventilation to cool the Arduino, CNC board, and stepper drivers is used.

**Grbl settings:**
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=6 (dir port invert mask:00000110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)
$101=400.000 (y, step/mm)
$102=400.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

Remark to settings $0 and $1, here I already tested some values (Ie. $0=5; $0=10; $0=25; $1=0; $1=25-50; $1=255). The results remain the same.

Puh, this got longer than expected, my apologies.
Thanks in advance to you for having a look,
Erik

blinkenlight, 2016-03-10:
I can't guarantee grbl has nothing to do with this, but are you sure this isn't mid-band resonance from your steppers? Are you using any micro-stepping (1/8 or better tends to get around that sort of thing)? Are your steppers running at their rated current?

Hogie1982, 2016-03-10:
@blinkenlight Thank you very much for your input. I just changed to 1/8 micro-stepping and this seems to have solved the problem. I will run further tests now. And will leave comment whether later about the results.
Again, thanks a lot.

Hogie1982, 2016-03-10:
Changing to 1/8 micro-step has solved the problem. So no Grbl, but the mid-resonace issue as @blinkenlight mentioned seemed to be the problem.

closed 2016-03-10 by Hogie1982