grbl/grbl #931: Level Translator for 3.3v to 5v

Moffy, 2016-03-10:
If GRBL moves to a newer ARM style board, some older systems will need the output levels boosted from 3.3v to 5v. I am including a possible solution.
[GRBL_3v3_to_5v.pdf](https://github.com/grbl/grbl/files/168127/GRBL_3v3_to_5v.pdf)
The circuit would use 3, 14pin IC's which are available in PDIP or non surface mount packages making it easy for the hobbyist to fabricate. The buffer IC's would do the level translating as well as providing drive for the signals. An extra signal is included 'STEP_CLK' which provides a 1us delay for the STEP_X_OUT etc signals. This would give a suitable setup time for the DIR_X_OUT etc signals to be applied before the step pulse is generated.
The idea is that the STEPPER timer interrupt would output all the step, dir and STEP_CLK at the same time and the hardware delay on the step pulses would provide a suitable setup time. The STEPPER timer interrupt would then, say 10us later reset the step and STEP_CLK pulses. This would slightly simplify the STEPPER timer interrupt. It would also mean that as the processors get faster less cycles would be wasted for the step setup time delay.
The circuit would also have headers so that the user could choose between the normal and inverted output, meaning that the invert mask could be removed also.

langwadt, 2016-03-11:
why add hardware to do something the code already does with a few lines of code?

109JB, 2016-03-11:
Why not just use simple logic level converters. That's what they are made for.

Protoneer, 2016-03-11:
Simple mosfet(level shifter) with appropriate pull up/down resistor. Will
handle any voltage and lots more current.

electrokean, 2016-03-11:
Be careful of the common single mosfet level shifters like the ones from Sparkfun or Adafruit.
These have reasonably slow slew rates (rise & fall times), and won’t handle fast step pulses very well.
You would need to use something that has active edge-rate acceleration like the TXB0104.

Protoneer, 2016-03-11:
Agreed. Not all mosfets are made equal so spec the part for the job.

electrokean, 2016-03-11:
I should clarify - the TXB0104 is probably overkill, as it is bi-directional which is partly why the edge acceleration comes into play.
Any uni-directional logic device that has reasonably fast propagation time and can handle the required logic level voltages will do.  For example many of the 74LVT/74LVC/74VHC devices when operating at 5V are properly specified to handle 3.3V logic level inputs.  Most of these are not available in DIP format though.

electrokean, 2016-03-11:
It isn't so much the mosfet specs that are the issue, but the use of pull up resistors.  You need a push-pull style output to get the fast rise & fall times.  Otherwise you're likely to have missed step pulses.

AlexHolden, 2016-03-11:
If you're using stepper drivers with opto-isolated inputs or an opto-isolated breakout board, you may simply need to swap the current limiting resistors to suit the lower drive voltage (check the GPIOs can handle the LED drive current though).

Moffy, 2016-03-11:
"why add hardware to do something the code already does with a few lines of code?"

Because sometimes hardware can do it better. The 1us delay for setup time is awkward if you want 1 us especially on the faster cpu's, you have to introduce a wait loop.This is burning cycles doing nothing when they could be doing something else.
If you use MOSFET level translators you are going to need to buffer them with say a 74HCT14, if you just want to buffer just use a 74HCT14 it will reliably convert from 3.3v to 5v while providing high drive current for driving the capacitance of a cable. No specific level translator required.

 "(check the GPIOs can handle the LED drive current though)."

Another good reason to use the 74HCT14. But if you are going to add a buffer, a few extra gates can add functionality. The STEP_CLK output is unnecessary if you use a 3 input NOR gate (74HCT27) with the 3 inputs connected to each of the STEP_X, Y, Z outputs, and its output connected to R1. These chips cost just 10s of cents each, and are readily available from multiple manufacturers.

109JB, 2016-03-11:
"If you use MOSFET level translators you are going to need to buffer them with say a 74HCT14,"

Why?  I have had contact with people using TinyG2 on Arduino Due and they reported that the cheapo logic level converters alone worked fine for interfacing to external stepper drivers.

Wouldn't another option would be to use high speed optocouplers to perform the level shift.?

Protoneer, 2016-03-11:
What speed signals can stepper drivers handle?

I had a look at A4988 drivers the other day and I think they could go up to
300Khz max... (anyone able to confirm that number?)

Moffy, 2016-03-11:
The 74HCT14 is effectively a 6 channel MOSFET level shifter(inverting) for 3.3v to 5v that also provides high input impedance / low output impedance, and requires no biasing components. It is generic, costs literally cents and available in a wide range of packages. Look at the data sheet, specifically the 74HCT version. It also provides hysteresis, which helps prevent oscillation and improves noise immunity.

109JB, 2016-03-11:
My CNC's use larger stepper drivers and the specs for most of these are on the order of 200-400 KHz max step rate. Not sure about the smaller drivers.

Moffy, 2016-03-13:
An improved version of the level translator/buffer that doesn't require a 'STEP_CLK' output. I use an NPN transistor and 4 resistors as a 3 input NOR gate.
[GRBL_3v3_to_5v.pdf](https://github.com/grbl/grbl/files/171265/GRBL_3v3_to_5v.pdf)
Also timing delay values are improved so it is more like 1us.

closed 2016-09-03 by chamnit