grbl/grbl #932: Command to return Grbl command queue state

casiobearing, 2016-03-12:
I'm just implementing a tool change mechanism in my CNC GUI that is driving my GRBL CNC machine.

When the GRBL is running full flow processing buffered GCode, a tool change is initiated in the GUI but the CNC machine is still busy working through the buffer of GCode prior to the tool change already sent to the GRBL.
I need to wait until the GRBL is idle before presenting the tool change request in the GUI.

I need a command to allow polling the GRBL to detect when it has finished processing its buffered GCode. i.e. a "Is Grbl Idle" command.

Any suggestions as to what command to use and what format the response should be... or if someone whats to build this into the next GRBL release?

Thanks.

cri-s, 2016-03-12:
G4p0
It OK is send after machine have done all motion.

casiobearing, 2016-03-13:
Hiya,

Thanks for the info... will try it out later

:o)

---- On Sat, 12 Mar 2016 15:08:08 +0000 cri-s<notifications@github.com> wrote ---- 
 > G4p0
 >  It OK is send after machine have done all motion.
 >

casiobearing, 2016-03-19:
Hi,
Just some feedback ... Had no luck using that command.... the best way I found to detect if the CNC machine has stopped moving is to examine the Planning buffer count ("Buf:" in the status messages)
if Planner count = 0, the machine is idle.

closed 2016-03-19 by casiobearing

chamnit, 2016-03-19:
G4 P0 or P with a very small value will empty the planner queue and then respond with an ok when the dwell is complete. At that instant, you have no queued motions, as long as you haven't sent more commands after the G4. This is the fastest way to do it without having to check the status reports.