grbl/grbl #933: Spindle On/off with m3/m5 has no effect until a speed is specified with VARIABLE_SPINDLE disabled

thibalte, 2016-03-17:
Hi there,

I just pulled my hairs for half a day trying to figure this one out, I have been using Grbl for a long time with pen plotters using M3/M5 to print with a solenoid, and after reading about the D12/D11 pin swap caused by VARIABLE_SPINDLE i commented it out from source, rebuilt and flashed. Still, Grbl will not toggle the ENABLE pin until a spindle speed is specified with the S command, even when it should not be needed since VARIABLE_SPINDLE is disabled. I believe M3/4 commands should be immediately executed when  speed control is disabled, right?

chamnit, 2016-03-17:
This may be the result of a recent bug fix. The gcode standard mandates that a spindle speed of zero (S0) disables the spindle. So an M3/4 will not turn it on unless S>0. 

I may have mandated this without variable spindle enabled, which doesn't make much sense. I'll look into it to see if I did so and will issue a fix, if it's there.

thibalte, 2016-03-17:
Great, thanks! Actually if the standard mandates it, maybe the best is to follow it and maybe update the wifi, notifying that even when disabling variable spindle, the enable/disable pin will not toggle unless a dummy speed above 0 is specified?

langwadt, 2016-03-17:
line 123 in spindle_control.c: if (rpm <= 0.0) { spindle_stop(); }

chamnit, 2016-03-18:
I pushed the fix for this issue last night. With VARIABLE_SPINDLE disabled, Grbl will ignore the S spindle speed value completely and turn on the spindle enable pin, regardless of what S is.

closed 2016-03-18 by chamnit

thibalte, 2016-03-18:
great! thank you very much.