grbl/grbl #937: Commands just work one

Kochchri, 2016-03-23:
Hi, 
I'm currently trying to get my software run. My shields and steppers are working fine, because if I just write a simple code to drive my shields with 2 wires each, they work. Also I get all the information when I use `$$` or other Commands. 

The strange thing is, that when I boot up, unlock my Arduino with `$X` and send `G1 X10 F1270`, the machine executes this command and my steppers move. Also I get the `ok` statement. But if I just want to do this again, nothing happens. I get again the `ok` but the steppers don't move. After a reset, exactly the same behaviour :(

jahnj0584, 2016-03-23:
You really should use all 4 wires.... 

Send a G91 command and try again. It should work now. Your machine was in g90 which is absolute position mode (if you sent x270 it goes 270 from zero. X271 after that would only move it one mm)

Kochchri, 2016-03-29:
you are right, thanks a lot. Its workin fine now. But I can not use the `$h` command. After I enter this command, nothing happens until the alarm fires `Homing failed`. I got two end-switches as pull-offs.

closed 2016-03-29 by Kochchri