grbl/grbl #938: Suggestion for next GRBL: Accel = f(speed), like stepper torque curve

dmalicky, 2016-03-24:
Here's an idea for a movement planner feature, to obtain both high acceleration and high rapid speed with stepper motors.  It's probably not a new idea, but I don't know of any hobby planners that do it.  

Background, as most of you probably know:
- CNC run times depend on axis acceleration and max speed, especially for intricate features, light cuts, and long rapids.  
- A stepper motor's torque is max at stall, decreasing to ~zero around 1000-2000 rpm.  
- Max axis acceleration is limited by motor torque (a = F/m).  
- Conventional movement planners ask the user to specify a maximum acceleration, but this acceleration is fixed, independent of motor RPM and axis velocity.  The user specifies a value that's a compromise of quick accel and top speed -- can't have both.  
- S-curve acceleration planners give smooth movement (no jerk), but don't account for the torque-RPM relationship of the motor.  And S-curve accel run times are typically slower than bang-bang accel, because of slow accel from a stop. 

Please see this pic to explain the idea:  [http://home.sandiego.edu/~dmalicky/accel.jpg](url)

The blue line is the torque curve of a typical stepper motor.

The red line is a typical constant acceleration setting:
- Accel value is chosen to give some % reserve torque at 1000 rpm.  
- Speed is limited to 1000 rpm, because above that is a high risk of losing steps.  
- At low rpm, there is a vast reserve of torque that is never used.   
- Of course, the user could choose a higher acceleration, but they'd have to further limit the rapid speed to still have enough reserve torque.   

The green line shows a linear acceleration profile:
- At 0 rpm, there is 2.5x more acceleration than the constant case, yet the stepper stays well within its torque capacity.  
- At 1000 rpm, accel is the same as the constant case.  
- Reserve torque is similar to the constant case (green line is ~parallel to blue).  
- The axis continues to accelerate to ~1500 rpm for better rapid speed.  And there is no jerk when the max speed is reached.
- A full S-curve accel planner could now utilize the higher torque at low rpm: with that, it could be faster than bang-bang accel.  
- So the user gets much quicker accel, faster rapids, and no jerk (at least at top speed).

I have no idea how hard this would be to implement.   I recall Sythetos' has S-curve accel on the Due cpu, so perhaps this is feasible on an ARM-based GRBL?

Thoughts?

chamnit, 2016-03-24:
@dmalicky : This is actually an idea I've been bouncing around for a while now. It's very true that you can increase machine performance by modeling the motor torque as something non-constant, or a linear slope approx, like you have shown. The issue is that the planning math becomes a nonlinear and non-trivial problem to solve in realtime. (TinyG's planner is not a true high order planner either. The last time I checked, it simply brute forces _a_ solution, but can't mathematically guarantee an optimal solution or a solution at all, which can lead to lost steps.) There is a reason why LinuxCNC or Mach3 don't have higher-order planning either. It's not as simple as it seems. Theres lots of research papers that look into this problem for industrial purposes.

Anyhow, this is something I continue to work on to see if a robust algorithm can be made that guarantees an optimal solution and never diverges. I think I'm close, but whenever I think that, I'm usually wrong.

dmalicky, 2016-03-25:
@chamnit   Thanks, Sonny, very interesting.  Good to hear you have already worked on it (and many other ideas, I'm sure).  Thanks for all your efforts on GRBL.  

I didn't know that about the TinyG planner.   I wondered if industrial machines needed this type of accel = f(rpm) control, since they use servos which are mostly constant torque.  

I can imagine the math for higher-order 3D planning is very complex.   I remember Art wrote the Tempest planner as an experiment for Mach3, and it worked very well, but was about 30% slower.  IIRC from trying it years ago, the user still spec'd 1 accel value (not accounting for the stepper torque curve) -- i.e., it was S-curve but only that.  

I completely agree the solution needs to be robust to not miss steps.  That trumps speed and smoothness.  Could you say more what is meant by "optimal"?  E.g., if there is a 2x gain in low speed stepper accel, even a substantial deviation from the optimal solution may be ok?  

I'm curious, if it were one or the other:
1. S-curve accel / minimal jerk control, for smoothness alone  (with no special accounting for torque = f(rpm) )
2. Accel = f(rpm) control, for speed alone  (with no special attempt at minimizing jerk)
Which would be harder to implement?  

Thanks, David

chamnit, 2016-03-25:
@dmalicky : Industrial machines usually have FPGAs or something much more powerful than an AVR or even ARM chips. To solve the higher-order motion control problem, they just use sheer power.

I mean optimal by Grbl's planner produces a mathematically optimal plan that minimizes time or maximize acceleration within the assumptions of the model and what it has in the buffer. Just as a note, whenever you add 3rd order jerk to a plan, you actually make the machine move slower, since the acceleration curves don't have as much area below them over the same amount of time. That is if the acceleration settings are the same. 

Of course, things can be done to create sub-optimal motion profiles, but I think it's not worth the effort to make something imperfect that will need to be constantly patched for certain motion scenarios. It's better to pose the problem and formally solve it for all cases. It might be impossible or very difficult, but I think it's worth trying.

The latter access control would be easier. The problem is better posed.

dmalicky, 2016-03-25:
@chamnit    Thank you, great stuff.   

That makes sense why jerk control is slower.  I just read up about Tempest -- near the end of its development, Art was able to get the run times to be about the same.  Like you mentioned, he simply increased the max accel by 30%, since he found the motors could handle that vs bang-bang accel.  I don't know if that strategy is supported by the math.    
http://www.machsupport.com/forum/index.php/topic,12042.msg96261.html#msg96261

That is awesome you are going for the optimal solution.  I agree if sub-optimal means the planner has to be patched just to execute certain motions, it's not a great solution.  I'm gathering there are (at least) these aims for a planner? 
1. Robust:  does not miss steps, can handle any motion scenario
2. Computationally feasible: for whatever hardware is assumed
3. Fast: high accel and rapids, short run times
4. Smooth: minimal jerk 
5. Optimal: fastest and/or smoothest possible solution
Not exactly a piece of cake.

chamnit, 2016-03-26:
@dmalicky : Interesting. Did not know Art tried implementing S-Curves, but yes, increasing the max acceleration would counteract the slower s-curve plans. 

Indeed, not exactly a piece of cake, but it isn't worth doing, if it isn't hard.

dmalicky, 2016-03-27:
@chamnit   Glad you are heading up GRBL.

Moffy, 2016-03-31:
[Stepper_Motor_Speed_Profile.pdf](https://github.com/grbl/grbl/files/196831/Stepper_Motor_Speed_Profile.pdf)
I assume some have read this pdf of stepper planning before. I have placed it here for those that haven't. David Austin, seems to be able to do it all while running on a PIC, but with some clever simplifying math.

chamnit, 2016-03-31:
@Moffy : Yes. This stepper algorithm has been around for a while. I played with it a while back. From what I remember, the main issue with it was numerical problems like round-off. It works only in a narrow window of frequency. And it has a ton of divides.

Moffy, 2016-03-31:
Good to know. Thanks.

closed 2016-04-14 by dmalicky