grbl/grbl #940: Grbl v0.9j incorrectly getting into Hold condition in normal operations

casiobearing, 2016-03-26:
Hi,
Here is a dump of Tx\Rx comms between my GUI and Grbl just after an initial tool change has completed and the GUI is resuming file machining operations...

For some reason Grbl is going into a STATE_HOLD condition.
Looking thru my comms dump I can see no feed hold "!" command being issued.
The start of each dump line is system time

The Grbl firmware v0.9j is compiled with the safety door option disabled (default setting) - i.e.
// #define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. 

Any ideas greatly appreciated...
Regards

G20
$#
?
00:00:00.0129360:'ok'
00:00:00.0129360:'[G54:30.000,30.000,-30.000]'
00:00:00.0129360:'[G55:0.000,0.000,0.000]'
00:00:00.0129360:'[G56:0.000,0.000,0.000]'
00:00:00.0129360:'[G57:0.000,0.000,0.000]'
00:00:00.0129360:'[G58:0.000,0.000,0.000]'
00:00:00.0129360:'[G59:0.000,0.000,0.000]'
00:00:00.0129360:'[G28:0.000,0.000,0.000]'
00:00:00.0129360:'[G30:0.000,0.000,0.000]'
00:00:00.0129360:'[G92:0.000,0.000,0.000]'
00:00:00.0129360:'[TLO:27.975]'
00:00:00.0129360:'[PRB:5.000,40.000,-49.768:1]'
00:00:00.0129360:'ok'
00:00:00.0129360:'<Idle,MPos:30.000,30.000,0.000,WPos:0.000,0.000,2.025,Buf:0,RX:0>'
?
00:00:00.0093144:'<Idle,MPos:30.000,30.000,0.000,WPos:0.000,0.000,2.025,Buf:0,RX:0>'
M3
00:00:00.0069762:'ok'
?
00:00:00.0080617:'<Idle,MPos:30.000,30.000,0.000,WPos:0.000,0.000,2.025,Buf:0,RX:0>'
?
00:00:00.0074552:'<Idle,MPos:30.000,30.000,0.000,WPos:0.000,0.000,2.025,Buf:0,RX:0>'
G0X1.072Y0.724
00:00:00.0084137:'ok'
?
00:00:00.0096328:'<Run,MPos:30.260,30.175,0.000,WPos:0.260,0.175,2.025,Buf:1,RX:0>'
?
00:00:00.0128288:'<Run,MPos:33.953,32.670,0.000,WPos:3.952,2.670,2.025,Buf:1,RX:0>'
G1Z-0.2F20
00:00:00.0071792:'ok'
?
00:00:00.0083079:'<Run,MPos:40.805,37.298,0.000,WPos:10.805,7.298,2.025,Buf:2,RX:0>'
?
00:00:00.0071340:'<Run,MPos:47.793,42.018,0.000,WPos:17.793,12.018,2.025,Buf:2,RX:0>'
G0Z0.25
?
00:00:00.0077799:'<Hold,MPos:53.950,46.175,0.000,WPos:23.950,16.175,2.025,Buf:2,RX:0>'
?
00:00:00.0066759:'<Hold,MPos:56.638,47.990,0.000,WPos:26.638,17.990,2.025,Buf:2,RX:0>'
G0X1.172Y0.724

casiobearing, 2016-03-26:
I just noticed pasting angle bracket status output from Grbl doesn't display in this wiki!
so here is the earlier comms dump output using [] instead of angle brackets...
G0X1.072Y0.724
00:00:00.0063519:'ok'
?
00:00:00.0064272:'[Run,MPos:30.045,30.030,0.000,WPos:0.045,0.030,2.070,Buf:1,RX:0]'
?
00:00:00.0086360:'[Run,MPos:33.185,32.150,0.000,WPos:3.185,2.150,2.070,Buf:1,RX:0]'
G1Z-0.2F200
00:00:00.0067279:'ok'
?
00:00:00.0098154:'[Run,MPos:39.623,36.498,0.000,WPos:9.623,6.498,2.070,Buf:2,RX:0]'
?
00:00:00.0077891:'[Run,MPos:46.375,41.060,0.000,WPos:16.375,11.060,2.070,Buf:2,RX:0]'
G0Z0.25
?
00:00:00.0092558:'[Hold,MPos:53.008,45.538,0.000,WPos:23.008,15.537,2.070,Buf:2,RX:0]'
?

Following more investigation I have found that if I added "G4 P1" after the "G1Z-0.2" commands, the error changes to be an expected "ALARM: Soft limit" error.
(It's an expected error because Grbl is trying to move Z up 0.25" which is past my upper Z limit home position of zero. My Z axis is 0 at Top, -57 at Bottom).

The Feed Hold status condition seems to be occurring due to the lack of the dwell between moving Z down -0.2" (a valid move which would leave MPos Z at -5mm, and the +0.25" which would end up with MPos Z at +1.35mm (which is above my soft Z limit of 0).

G0X1.072Y0.724
00:00:00.0054218:'ok'
?
00:00:00.0096743:'[Run,MPos:31.343,30.908,0.000,WPos:1.343,0.907,2.070,Buf:1,RX:0]'
?
00:00:00.0064416:'[Run,MPos:36.565,34.433,0.000,WPos:6.565,4.432,2.070,Buf:1,RX:0]'
G1Z-0.2F200
00:00:00.0079432:'ok'
?
00:00:00.0082353:'[Run,MPos:43.725,39.270,0.000,WPos:13.725,9.270,2.070,Buf:2,RX:0]'
G4P1
?
00:00:00.0083285:'[Run,MPos:50.400,43.778,0.000,WPos:20.400,13.778,2.070,Buf:2,RX:0]'
?
00:00:00.0088945:'[Run,MPos:56.023,47.575,0.000,WPos:26.023,17.575,2.070,Buf:2,RX:0]'
G0Z0.25
?
00:00:00.0101503:'[Run,MPos:57.230,48.390,-0.275,WPos:27.230,18.390,1.795,Buf:1,RX:8]'
?
00:00:00.0093811:'[Run,MPos:57.230,48.390,-3.423,WPos:27.230,18.390,-1.353,Buf:1,RX:8]'
G0X1.172Y0.724
?
00:00:00.0071676:'[Run,MPos:57.230,48.390,-6.605,WPos:27.230,18.390,-4.535,Buf:1,RX:23]'
?
00:00:00.0132860:'[Idle,MPos:57.230,48.390,-7.150,WPos:27.230,18.390,-5.080,Buf:0,RX:23]'
G1Z-0.2
?
00:00:00.0145013:'[Idle,MPos:57.230,48.390,-7.150,WPos:27.230,18.390,-5.080,Buf:0,RX:31]'
G4P1
?
00:00:00.0515590:'[Idle,MPos:57.230,48.390,-7.150,WPos:27.230,18.390,-5.080,Buf:0,RX:36]'
?
00:00:00.0532582:'[Idle,MPos:57.230,48.390,-7.150,WPos:27.230,18.390,-5.080,Buf:0,RX:36]'
G0Z0.25
?
00:00:00.0298318:'[Idle,MPos:57.230,48.390,-7.150,WPos:27.230,18.390,-5.080,Buf:0,RX:44]'
00:00:00.0298318:'ok'
00:00:00.0298318:'ALARM: Soft limit'

chamnit, 2016-03-26:
@casiobearing : This problem has been fixed in the most recent version of Grbl. Please try that.

closed 2016-03-26 by chamnit

casiobearing, 2016-03-27:
> &mdash;

Hiya,

I've just updated Grbl and can confirm the Hold issue is fixed.

Can I recommend updating the Grbl version (or perhaps append\use a 3 digit build number) whenever any changes are made as it wasn't clear on github there was in fact a revised version of 0.9j.

Regards
Ian

---- On Sat, 26 Mar 2016 18:26:02 +0000 Sonny Jeon  wrote ----

chamnit, 2016-03-27:
There is one. Issue Grbl a $I command and it'll tell you the version date.

casiobearing, 2016-03-28:
Hi,

Didn't know that - thanks... but it would be useful if that was also written on the website\github somewhere too.

Regards
Ian 

---- On Sun, 27 Mar 2016 13:39:39 +0000 Sonny Jeon  wrote ----