grbl/grbl #943: Hard limit appearing random

Kochchri, 2016-03-29:
I'm using grbl 0.9 and the grbl controller 3.6.1. When I start homing, it seems to work correctrly except of the homing is done twice ( I'm using a 2 axis machine). But when I want to use the Axis Control to drie my axis, after at least one step, a Hard-Limit-Alarm is fired. But the limit-switcher are not triggered. And they work, as the homing shows. I also read both of my switches with an simple arduino sketch which showed that they work correctly wothout any distursion.

chamnit, 2016-03-29:
@Kochchri : This is a very common issue and the fix is to limit electrical interference in the limit pins. Wire in a low-pass filter RC circuit on each pin, shield the limit cables, place them away from any powered wires, etc,etc.

closed 2016-03-29 by chamnit

Kochchri, 2016-03-30:
I solved it by soldering in a .47µf capacitor between GND and my switches. Workes fine.